ixmilia/dxf #24: Add support for remaining entity types

brettfo, 2015-06-09:
* HATCH
* HELIX
* LIGHT
* MESH
* MLINE
* MLEADER
* MLEADERSTYLE
* MTEXT
* SECTION
* SURFACE
* TABLE
* UNDERLAY
* VIEWPORT

arendvw, 2017-01-11:
Is there any way I can assist in these issues? 

Hatch also has my interest to have supported. (Recognizing the boundaries would already be enough for me)

Kabeja seems to have a working implementation partially described here, is it an idea to 
https://github.com/fuzziness/kabeja/blob/f126b44ac32962004217be53bfca59fd989fcb95/blocks/dxf/src/main/java/org/kabeja/dxf/parser/entities/DXFHatchHandler.java

brettfo, 2017-01-11:
These entities are mainly ones that will require a custom reader (which I've done with other entities), but they will require much more work to get functioning properly.  E.g., the HATCH entity is particularly nasty because it contains many boundary paths, with each boundary path containing soft pointers to many boundary shapes.  A HATCH entity _also_ contains many pattern data segments.

DJGosnell, 2017-06-08:
I hate to bump such an old issue, but this project seems to be exactly what I am looking for.  I would like to contribute to the project in respect to working on the implementation of the Table entity.  Where would you suggest starting if I were to add to this project?

brettfo, 2017-06-08:
I haven't attempted to support the `TABLE` entity primarily because I've never used one in my limited experience with AutoCAD so I've never had a 'real' DXF file to compare to and [the spec (page 154)](http://images.autodesk.com/adsk/files/autocad_2014_pdf_dxf_reference_enu.pdf) doesn't give any information about how it's really used; it simply gives a lot of code-pair value meanings.

I'd love some help working on the `TABLE` entity and there are primarily two ways to go about it:

1. Auto-generate the code for it by adding an entry in the XML file [here](https://github.com/IxMilia/Dxf/blob/e42a18e1c1754f3d961f1fdddc17dac94fbae385/src/IxMilia.Dxf.Generator/Specs/EntitiesSpec.xml#L1054).
2. Manually write all of the code and only specify the properties like is done for [`MTEXT`](https://github.com/IxMilia/Dxf/blob/e42a18e1c1754f3d961f1fdddc17dac94fbae385/src/IxMilia.Dxf.Generator/Specs/EntitiesSpec.xml#L707).

wtaisto, 2020-05-06:
Is there any chance that limited HATCH support could be added that would just make it so I can get the "fill" color of LWPolyLine entities which have a hatch applied?  My application just needs to be able to look up the DxfColor.Index of the fill.

brettfo, 2020-05-07:
@wtaisto I can start looking into adding simple HATCH support.  Do you have an existing file with a valid HATCH entity that I could test with?  If you don't want the file made public you can email it directly to me at brett.forsgren@outlook.com and I'll delete it as soon as I'm done with it.

brettfo, 2020-05-09:
@wtaisto I've pushed a branch `hatch` that has some initial support for reading/writing HATCH entities.  Can you give that a shot and let me know if it does what you need?  If you're just trying to get the fill color, that's probably the `.Color` property that's common on every entity (now including HATCH, too).

wtaisto, 2020-05-10:
@brettfo Ok so I loaded the sample file I sent you using your hatch branch.  The hatch entity is recognized, and the color does get loaded correctly.  But I don't seem to see any geometry data in the object; the hatch.BoundaryPaths has a single item, but it contains no Edges.  There were a couple of other properties that seemed like geometry candidates based on the names, but I didn't see anything which appeared to have path-like data.

Where should I be looking for the geometry of the hatch border, or is there perhaps a single insertion point value, or something that would provide location?

Thank you SO MUCH for looking at this, I really appreciate it!

brettfo, 2020-05-10:
I think I figured out this issue; I misunderstood the meaning of a field.  With the latest commit to the `hatch` branch I can get this:

``` csharp
var file = DxfFile.Load(@"path\to\file\you\sent\me.dxf");
var hatch = file.Entities.OfType<DxfHatch>().First();
var boundaryPath = hatch.BoundaryPaths.First() as DxfHatch.PolylineBoundaryPath;
var boundaryPoints = boundaryPath.Vertices.Select(v => v.Location);
// at this point we have 4 boundary points that specify its location in the drawing
```

wtaisto, 2020-05-11:
Perfect!  The boundary points come out exactly where I'd expect them to.

brettfo, 2020-05-11:
Glad to hear it works, I've merged those changes to `master`.

wtaisto, 2020-05-11:
Awesome.  Do you have a release cycle for your NuGet package, or will the changes appear there soon?

brettfo, 2020-05-11:
I don't have any sort of release cycle, I just publish whenever there have been enough changes to warrant an update and when I get around to it.

If a new release of the package would be useful to you, I can certainly publish another version.