ixmilia/dxf #36: Do not rely on current culture when parsing double values

nzain, 2015-10-15:
First off: nice code. Looks professional!

However, I was scratching my head, when your library threw an ArgumentOutOfRangeException while parsing the following header part:

    $TDCREATE
     40
    2456478.590142998

Diving deep into your code revealed that you rely on double.TryParse(string s, out double d) which is culture dependent. I'm from germany - decimal separator is comma, thousands separator is dot. Thus, double.TryParse succeeds since it ignores the dot (as if it was a thousands separator) and returns the double value

    2456478590142998.0

Now add this to the julian calendar base and someone is scratching his head :-)

I'm not sure what the DXF spec says about decimal separators, but if it is culture invariant, you should use

    double.TryParse(s, NumberStyles.Float, CultureInfo.InvariantCulture, out result)

brettfo, 2015-10-15:
Thanks for the bug report!  You should be able to pull latest and rebuild.

closed 2015-10-15 by brettfo

nzain, 2015-10-16:
Thanks for the very fast response & fix. My files are loaded correctly now.

PR #37 (2015-10-15): use InvariantCulture when parsing ASCII files
Fixes #36.