ixmilia/dxf #54: Create Dxf files with R12

yigido, 2015-12-04:
How can I create Dxf files with version R12?
When I set the version like below, no ather Dxf Viewer can open the file.

> DxfFile oDxfFile = new DxfFile();
> oDxfFile.Header.Version = DxfAcadVersion.R12;
> DxfLine oLine = new DxfLine();
> oLine.P1.X = X1;
> oLine.P1.Y = Y1;
> oLine.P2.X = X2;
> oLine.P2.Y = Y2;
> oDxfFile.Entities.Add(oLine);
> ...
> ...

brettfo, 2015-12-04:
One of my backlog items is to improve readability by other viewers libraries (see #17), but this is a bit of a crap shoot since the DXF spec is so loosely defined and implemented.  What other viewers/libraries are you using and what do the errors say?  I'd love to make this work universally if possible.

For what it's worth, you're using this code exactly like I intend it to work (by setting `oDxfFile.Header.Version = DxfAcadVersion.R12`.  Also, does it work if you set the version to `DxfAcadVersion.R10`?

yigido, 2015-12-08:
I write a .Net editor for R12 files, to open, edit and save.
I tried to open the saved files with AutoCAD 2016 but it doesn't work and i see no error message.

Autodesk online viewer shown (https://a360.autodesk.com/viewer)
>'Sorry, the drawing file is invalid and cannot be viewed. - Please try to recover the file in AutoCAD, and upload it again to view.'

A free AutoCAD viewer shown (https://gallery.proficad.com/tools/AutoCAD-Viewer.aspx)
>'we could not open this drawing'

I tested oDxfFile.Header.Version = DxfAcadVersion.R12 and oDxfFile.Header.Version = DxfAcadVersion.R10 but not work.
The DXF-file shows objects witch are not supported from the version I set.

brettfo, 2015-12-28:
I've recently been working on improving compatibility and you may have some luck with the `autocad-compat` branch when using `file.Header.Version = DxfAcadVersion.R12`.

PR #57 (2016-05-19): Improve compatibility with other readers/writers
Improves compat with other readers and writers, specifically the Teigha libraries from the Open Design Alliance and AutoCAD.

Fixes #17.
Fixes #54.
Fixes #56.
closed 2016-05-19 by brettfo