ixmilia/dxf #55: Help extracting this information

rhyek, 2015-12-15:
Hey guys. This is sort of a long shot, but I was wondering if anyone could give me some guidance into how I could be able to extract the information in the table shown in this [image](http://imgur.com/OhCzyxf).

I have a possible small project where I need to be able to grab a DXF file representing architecture blueprints made with AutoCAD, find one or several of those tables and convert the data to human-friendly descriptions.

I've used a couple of DXF viewers to be able to understand their structure, but to my untrained eye there doesn't seem to be any. The file seems to be composed of visual elements such as lines and labels with no real relationship between them.

Any ideas? Thanks.

brettfo, 2015-12-15:
The DXF format doesn't expose anything like what you're looking for and I'm guessing that table is just a bunch of TEXT and LINE entities that just so happen to align nicely.

Depending on how nicely the TEXT entities align, you might be able to divine the pattern yourself.  The below code certainly won't work, but it might give you some ideas for a starting point.

``` C#
var textEntities = dxfFile.Entities.OfType<DxfText>(); // all TEXT entities in the file
var textYValues = new Dictionary<double, List<DxfText>>(); // a collection of all TEXT entities with the same Y value
foreach (var text in textEntities)
{
    if (!textYValues.ContainsKey(text.Location.Y);
    {
        textYValues.Add(text.Location.Y, new List<DxfText>());
    }

    textYValues[text.Location.Y].Add(text);
}
```

Then once you've done this, try to correlate the y-values with something that might be a table.

``` C#
// in your sample image, the table has 4 columns
foreach (var kvp in textYValues.Where(l => l.Count == 4))
{
    // the y-value of what we think is a row in the table
    var yValue = kvp.Key;

    // a list of what we think are the values for the given row, sorted left-to-right
    var columnValues = kvp.Value.OrderBy(v => v.Location.X);

    // do something with them
}
```

This approach has a few notable problems, though:
1.  The location of items in the file are specified by `double`s and it's unlikely that all of the items on a row are going to have the exact same y-value.  E.g., one item might be `100.0` and another might be `100.000000000001`.  You could potentially get around this by bucketing the TEXT entities based on some slightly truncated value, say rounded to 6 decimal places?
2.  One row of the table might have empty cells which means that row wouldn't get picked up.   A potential solution to this could be to detect that rows are semi-regular (e.g., every 20 units) and find the entities that way.
3.  Similar to (2), there might be other TEXT entities that aren't part of the table that just so happen to have a y-value exactly equal to other table items that could make that row look like it has more entities than the others.

Another though I had on how to filter the TEXT entities to the few that you want would be to do a statistical sampling of the `(x, y)` values of the locations and find a cluster of them and then limit your search for columns/rows to those.  This could weed out random notes strewn throughout the drawing.

rhyek, 2015-12-15:
Amazing answer. Thank you for taking the time. I believe what I'm looking for will only be found in the original AutoCAD project file (DWG). I hadn't understood entirely what DXF was intended for before asking this question.

Again, thanks.

closed 2015-12-15 by rhyek

brettfo, 2015-12-16:
There's still the possibility that even in the original DWG the table you're looking for will be just a collection of lines and text that just happen to look like a table, but in reading through the only copy of a DWG spec that I could find ([here](http://www.opendesign.com/files/guestdownloads/OpenDesign_Specification_for_.dwg_files.pdf)) there does appear to be an object of type `ACAD_TABLE` that might be what you're looking for.

Hope this helps.

rhyek, 2015-12-16:
Thanks. I'll have to run some tests by the weekend. I do know that the table is generated within AutoCAD from a drawn blueprint and its data so you may have a point in that the table itself is only lines and labels with no structure.

Hopefully there is some relationship between the labels and the table, or maybe I could be able to read the blueprint data and be able to generate what I need from that bypassing the table altogether (obviously not ideal or easy).