ixmilia/dxf #56: Failed to read old DXF file

nzain, 2016-01-12:
I just upgraded my NuGet package to 0.3.1 and now all files fail with the same exception. When debugging, I get about 10 assertion errors before... I'm unable to track it down. Exception stacktrace, debug trace, and example file follow.

IxMilia.Dxf.DxfReadException: Unexpected code pair group text
    at IxMilia.Dxf.DxfCodePairGroup.FromBuffer(DxfCodePairBufferReader buffer, String groupName) in DxfCodePairGroup.cs: line 86
   at IxMilia.Dxf.Objects.DxfObject.TrySetExtensionData(DxfCodePair pair, DxfCodePairBufferReader buffer) in DxfObject.cs: line 136
   at IxMilia.Dxf.Objects.DxfXRecordObject.PopulateFromBuffer(DxfCodePairBufferReader buffer) in DxfXRecordObject.cs: line 30
   at IxMilia.Dxf.Objects.DxfObject.FromBuffer(DxfCodePairBufferReader buffer) in DxfObjectGenerated.cs: line 314
   at IxMilia.Dxf.Sections.DxfObjectsSection.ObjectsSectionFromBuffer(DxfCodePairBufferReader buffer) in DxfObjectsSection.cs: line 43
   at IxMilia.Dxf.Sections.DxfSection.FromBuffer(DxfCodePairBufferReader buffer, DxfAcadVersion version) in DxfSection.cs: line 73
   at IxMilia.Dxf.DxfFile.LoadFromReader(IDxfCodePairReader reader) in DxfFile.cs: line 212
   at IxMilia.Dxf.DxfFile.Load(Stream stream) in DxfFile.cs: line 149
   at IxMilia.Dxf.Test.UnitTest1.Load(String filename) in UnitTest1.cs: line 31

Debug Trace:
Fail: 
Fail: Unexpected end of section/file while parsing group.
Fail: 
Fail: Unexpected end of section/file while parsing group.
Fail: 
Fail: Unexpected end of section/file while parsing group.
Fail: 
Fail: 
Fail: 
Fail: 

This is a dxf file (not supported by github):
[Clearance.txt](https://github.com/IxMilia/Dxf/files/87260/Clearance.txt)

brettfo, 2016-01-12:
Thanks for attaching a file, this made investigations much easier.

In 0.3.1 I have some bugs with reading extension data that I've recently been working on and you're hitting with this file, but haven't quite finalized this release yet.  If you can take a source dependency and build locally (instead of the NuGet package), the [`autocad-compat`](https://github.com/IxMilia/Dxf/tree/autocad-compat) branch contains a lot of fixes for compatibility issues I had between AutoCAD and this library and I've confirmed that I can open your file with this branch.

nzain, 2016-01-12:
I've built a custom version already (based on 0.2) that was working 
flawlessly. I think.. hope, that version is sufficient for 99% use cases.

I can only update if updates are stable (no offense), since we're close 
to release. No need to press additional functionality, if you're not ready.

brettfo, 2016-01-12:
No offense taken, I understand the need for stability.  That being said, if your custom version based off of 0.2 is working then I'd stay with that as I'm not yet ready to declare the `autocad-compat` branch ready to merge into `master` yet.

On a related note, if you're willing to share I'd like to see/merge your changes if I haven't already addressed the issues in the `autocad-compat` branch, but I also understand if you'd like to keep your changes private for either personal or business reasons.

nzain, 2016-01-12:
I didn't have any changes, I just pulled your latest version including 
fixes for one of my earlier problems - you just didn't publish a nuget, 
that's why I built my own. Currently, I have no need for extended 
functionality - this may change in the future, when customers start 
loading strange files :) The file attached to this report is a standard 
use case for us. As long as no special use cases arise, I'll stay with 
my self-built version. From time to time, I'll follow your releases, 
evaluate and check back.

PR #57 (2016-05-19): Improve compatibility with other readers/writers
Improves compat with other readers and writers, specifically the Teigha libraries from the Open Design Alliance and AutoCAD.

Fixes #17.
Fixes #54.
Fixes #56.
closed 2016-05-19 by brettfo