ixmilia/dxf #60: Enable AutoCAD to open R13 files generated by this library

brettfo, 2016-05-20:


nzain, 2020-01-31:
Actually, we emit version `R13` with

- some extra layers
- lines
- lwpolylines
- circles
- arcs
- aligned dimensions (almost working as intended)
- block+blockrecord+insert (especially important for "world coordinates" that require more than 32bit)

*and* AutoCAD 2017, 2019, 2020 do not crash. AutoCAD even displays almost all of my drawings.

The only thing I couldn't find out is how to make a dimension _inside_ a block display in this beast of a software. It doesn't crash, does not show an error (that helped me to fix all that other stuff), but it doesn't show the dimensions inside blocks...

Your milage may vary, but one could say "AutoCAD can open R13 files generated by this library".

nzain, 2020-01-31:
I have to add that we have some sophisticaed 3D exports as well. The DXF spec is terrible in this respect (draw 2D stuff and use a 3D normal vector to rotate it), but it works.

brettfo, 2020-01-31:
Thanks for all your help in finding and fixing the issues!  I'll leave this issue open for now so it can be a gathering place for any future compat issues that come up.