ixmilia/dxf #61: Can't open 2D export from Google Sketchup

AleksandarDev, 2016-05-26:
I'm having trouble opening DXF exported (2D) from Google Sketchup 2016.

```
IxMilia.Dxf.DxfReadException was unhandled by user code
  HResult=-2146233088
  Message=Unexpected code pair group text
  Offset=3399
  Source=IxMilia.Dxf
  StackTrace:
       at IxMilia.Dxf.DxfCodePairGroup.FromBuffer(DxfCodePairBufferReader buffer, String groupName) in E:\IxMilia\Dxf\src\IxMilia.Dxf\DxfCodePairGroup.cs:line 86
       at IxMilia.Dxf.Objects.DxfObject.TrySetExtensionData(DxfCodePair pair, DxfCodePairBufferReader buffer) in E:\IxMilia\Dxf\src\IxMilia.Dxf\Objects\DxfObject.cs:line 135
       at IxMilia.Dxf.Objects.DxfXRecordObject.PopulateFromBuffer(DxfCodePairBufferReader buffer) in E:\IxMilia\Dxf\src\IxMilia.Dxf\Objects\DxfXRecordObject.cs:line 30
       at IxMilia.Dxf.Objects.DxfObject.FromBuffer(DxfCodePairBufferReader buffer) in E:\IxMilia\Dxf\src\IxMilia.Dxf\Objects\DxfObjectGenerated.cs:line 314
       at IxMilia.Dxf.Sections.DxfObjectsSection.ObjectsSectionFromBuffer(DxfCodePairBufferReader buffer) in E:\IxMilia\Dxf\src\IxMilia.Dxf\Sections\DxfObjectsSection.cs:line 43
       at IxMilia.Dxf.Sections.DxfSection.FromBuffer(DxfCodePairBufferReader buffer, DxfAcadVersion version) in E:\IxMilia\Dxf\src\IxMilia.Dxf\Sections\DxfSection.cs:line 70
       at IxMilia.Dxf.DxfFile.LoadFromReader(IDxfCodePairReader reader) in E:\IxMilia\Dxf\src\IxMilia.Dxf\DxfFile.cs:line 212
       at IxMilia.Dxf.DxfFile.Load(Stream stream) in E:\IxMilia\Dxf\src\IxMilia.Dxf\DxfFile.cs:line 144
       at FloorFlood.---.OpenDxFiles(IEnumerable`1 filePaths)
```

Lines 3399 to 3419 of DXF file
```
102
VTR_0.000_0.000_1.000_1.000_GRIDDISPLAY
 70
     3
102
VTR_0.000_0.000_1.000_1.000_GRIDMAJOR
 70
     5
102
VTR_0.000_0.000_1.000_1.000_DEFAULTLIGHTING
280
     1
102
VTR_0.000_0.000_1.000_1.000_DEFAULTLIGHTINGTYPE
 70
     1
102
VTR_0.000_0.000_1.000_1.000_BRIGHTNESS
141
0.0
102
VTR_0.000_0.000_1.000_1.000_CONTRAST
142
0.0
  0
```

brettfo, 2016-05-26:
Thank you for the bug report.  This looks like another case where the Autodesk DXF spec isn't complete; the spec says `102` codes are supposed to be of the format:

```
102
{some-string
// a bunch of other code pairs
102
}
```

But this looks like a specialized shortened version used when only one code pair would be in the group.

I have two questions for you:

1. How were you using this library?  NuGet package or clone/build locally?  What version/commit?
2. Can you email me a file that leads to this error?  **brett.forsgren<super></super>@outlook.com**

AleksandarDev, 2016-05-26:
Figured about the 102{ stuff from source code, but I'm not familiar with DXF spec. If this VTR thing is uncommon, I can manage without you changing the code. I can just delete those 6 parameters from the file. I tried that and now it works.

I'm using online NuGet version 0.3.1.

Here are both Sketchup file and DXF export file.

[FloorSkpAndDxf.zip](https://github.com/IxMilia/Dxf/files/285021/FloorSkpAndDxf.zip)

brettfo, 2016-05-26:
FYI, I've created PR #62 which will allow this file to open.

My original guess about the `102` handling was off and the problem was ultimately that the `XRECORD` object in your file didn't specify the duplicate record handling flag which means a subsequent read of the code `280` pair was mis-interpreted as that which caused the rest of the object to be read incorrectly.

closed 2016-05-26 by brettfo

AleksandarDev, 2016-05-26:
@brettfo Thanks for the quick response and fix!

PR #62 (2016-05-26): fix reading XRECORD objects with no duplicate-handling flag
When reading and `XRECORD` object, the only possibilities following the `100/AcDbXrecord` code pair are either a code `280` which indicates the duplicate record handling flag, or a code `102` which is just regular data.

This moves the code `280` handler higher in the loop to account for the case where a code `102` immediately follows the `100/AcDbXrecord` pair which means any subsequent code `280` must belong to the data section and not the duplicate record handling flag.

Fixes #61

FYI @AleksandarDev