ixmilia/dxf #63: Is there any way to fill color in more than four vertexes or less than four vertexes polyline area?

idazhenhan, 2016-08-19:
I am using DxfSolid to fill color for a polyline entity, I saw it supports just four vertexes.
If a polyline entity has three or five(or much more) vertexes, the DxfSolid  setting will be invalid.
Can you show me some examples of how to fill a  polyline with custom color in right way?

Thank you!

brettfo, 2016-08-19:
Admittedly I'm not very familiar with the `SOLID` entity type (represented by `DxfSolid` in this code base) but according to the official AutoCAD spec [here (page 152)](http://images.autodesk.com/adsk/files/autocad_2014_pdf_dxf_reference_enu.pdf) (PDF warning), `SOLID` entities have exactly 4 vertices and you'd have to use a polyline for less/more.

idazhenhan, 2016-08-20:
Thank you for your response.. It seems I have to find another way to fullfill my needs, the spec may be helpful :]

idazhenhan, 2016-08-20:
Sorry for the reopen operate.

Is there have a plan to support for [Dxf HATCH(page 98)](http://101.96.10.70/images.autodesk.com/adsk/files/autocad_2014_pdf_dxf_reference_enu.pdf) and [Dxf Boundary Path Data(page 101)
](http://101.96.10.70/images.autodesk.com/adsk/files/autocad_2014_pdf_dxf_reference_enu.pdf)?, I found these two parts are very useful to resolve problems like my needs.

brettfo, 2016-08-22:
I do eventually want to support `HATCH` entity types as part of #24 (and the corresponding boundary path and pattern data), but it's a complex enough entity that I don't see myself getting around to it any time soon.

idazhenhan, 2016-08-23:
Get it. Thanks!

closed 2016-08-23 by idazhenhan