ixmilia/dxf #66: Wipeout can not be readed in AutoCAD2016

whuaegeanse, 2017-04-10:


brettfo, 2017-04-10:
Can you provide some more detail?  Is the problem that AutoCAD can't open a `WIPEOUT` entity created with this library, or that this library can't read a `WIPEOUT` entity created by AutoCAD?  If it's the former can you provide a code sample, and if it's the latter do you have a file you could share that I could test with?  Are you using the NuGet packages or building from source?

whuaegeanse, 2017-04-12:
I use this library by building from source myself with .net 4.0. 
The code is pasted as follow:
`        static void TestWipeout()
        {
            string fileIn = @"E:\test1.dxf";
            if (System.IO.File.Exists(fileIn))
            {
                System.IO.File.Delete(fileIn);
            }
            DxfFile dxfFile = new DxfFile();
            DxfWipeout pDxfWipeout = new DxfWipeout();
            pDxfWipeout.ClippingVertices.Add(new DxfPoint(-30, 30, 0));
            pDxfWipeout.ClippingVertices.Add(new DxfPoint(-20, 60, 0));
            pDxfWipeout.ClippingVertices.Add(new DxfPoint(10, 70, 0));
            pDxfWipeout.ClippingVertices.Add(new DxfPoint(-10, 40, 0));
            pDxfWipeout.ClippingVertices.Add(new DxfPoint(30, 20, 0));
            pDxfWipeout.ClippingType = DxfImageClippingBoundaryType.Polygonal;
            pDxfWipeout.UseClippingBoundary = true;
            dxfFile.Entities.Add(pDxfWipeout);
            using (FileStream fs = new FileStream(fileIn, FileMode.CreateNew))
            {
                dxfFile.Header.Version = DxfAcadVersion.R2013;
                dxfFile.Save(fs);
            }
        }

        static void Main(string[] args)
        {
            TestWipeout();
        }`
After loading the text1.dxf file into AutoCAD2016, there is nothing showed.

brettfo, 2017-04-12:
The `WIPEOUT` entity wholly inherits from `IMAGE` which requires that the `.ImageDefinition` property be set.  Something like this should work:

``` C#
using IxMilia.Dxf.Objects;
// ...
// assuming you've already set the clipping like above
pDxfWipeout.ImageDefinition = new DxfImageDefinition()
{
    FilePath = @"C:\path\to\file.jpg",
    ImageWidth = 640, // this should come from the image
    ImageHeight = 480, // this should come from the image
    PixelWidth = 1.0, // I'm not sure if this should be 0.0, 1.0, or ImageWidth
    PixelHeight = 1.0, // same concern here
};
```

BKSpurgeon, 2018-03-07:
If wipeout needs the Imagine definition set - what about requiring that to be passed into the constructor? I could make the change and put in a pull request?

brettfo, 2018-03-07:
I recently (~13 days ago) made `DxfImage` easier to instantiate correctly in commit a48cba6d4a107c0c39ee4d08730d65b94877f86c so I just now in commit 49151a1f3b0cc18c62ae1c17b2028102848a19bc added a similar constructor to `DxfWipeout`.

I don't have a copy of AutoCAD to test with, but both [QCAD](https://qcad.org) and the [Teigha Viewer](https://www.opendesign.com/guestfiles/teigha_viewer) from the Open Design Alliance can display an `IMAGE` entity created with this library now (which means they should handle `WIPEOUT`, too, since they're identical.)

closed 2018-07-09 by brettfo