ixmilia/dxf #68: How to deal with Units?

nzain, 2017-04-18:
I realized, that dxf files were unitless (once upon a time) and units were something the drawing application would add on top. Furthermore, I could not find version-independent (and human-readable) instructions how to deal with units in DXF files. DXF versions aside, this packages provides a great abstraction level and I was able to find several properties related to units. Problem is, I'm still not sure if my understanding of those properties makes sense. Please confirm or correct me, if I'm wrong.

**Preface: Units**
There are distance units (meters, millimeters, feet, inch, etc.) and angular units (rad, degree, etc.).
The metric distance system is unique, while the imperial system is not. There are (too) many definitions of a feet, in particular I have to deal with "international feet" (common) and "US survey feet" (still in use by many surveyors).
Additionally, applications may round numbers (e.g. 3 decimal places) to improve readability - lets skip this.

**Reading Units from a DXF File**
When reading values from a DXF file, you want to convert "their" units to your internal unit system, whatever that may be. Suppose I want all distance values in meters, then the following method would provide the conversion factor. Same could be done for angles.

    static double GetDistanceConversionFactor(DxfHeader dxfHeader)
    {
        switch (dxfHeader.DefaultDrawingUnits)
        {
            case DxfUnits.Unitless: // assume meters
            case DxfUnits.Meters:
                return 1.0;
            case DxfUnits.Millimeters:
                return 0.001;
            case DxfUnits.Feet:
                return 0.3048; // only valid for international feet
            case DxfUnits.Inches:
                return 0.3048 / 12; // only valid for international feet
            default:
                throw new NotSupportedException("DefaultDrawingUnits=" + dxfHeader.DefaultDrawingUnits);
        }
    }

The DXF standard does not provide a way to distinguish international feet and US survey feet. While there is a property ´DxfHeader.DrawingUnits´ which distinguishes "metric" and "english", I don't know what this is good for. I think, I have to ask the user for his preferred system in case of feet/inch.

**Writing Values+Unit to DXF**
Essentially, this is the opposite direction - convert all values to the target unit before creating the entities. Additionally, set DxfHeader.DefaultDrawingUnits to that unit.

    DxfFile dxfFile = new DxfFile();
    dxfFile.Header.DefaultDrawingUnits = DxfUnits.Feet;
    dxfFile.Header.AngleUnitFormat = DxfAngleFormat.Radians;
    dxfFile.Entities.Add( ... ) // use feet/rad here
    dxfFile.Save(...);

Anything else, I have to do before saving? Maybe some property to complement the units stuff?

If my understanding of units in DXF files and your library is correct, feel free to reuse this for the main page or a wiki - so others can benefit.

brettfo, 2017-04-18:
A couple of points:

- **Units in General** - You are correct in that units were added as an afterthought and that you (the developer) will have to convert between the drawing's units and whatever you want to use internally.  This is a common theme in other CAD file formats as well.

- **Feet** - There's no way to appropriately distinguish between international and survey feet, so I think prompting the user is the only real way forward without making broad assumptions.  If you (or anybody else) comes across guidance on how to handle this, I'd love to incorporate it.

- **Angles** - You're not going to like what I have to say here, but it's the way it is: as I interpreted the spec, the `AngleUnitFormat` property is used by AutoCAD for how to display units to the user in `DIMENSION` entities, but the internal representation is fixed and non-standard.  E.g., `ARC` entities always specify their start and end angle values in degrees (_page 76_ of the [latest specification](http://images.autodesk.com/adsk/files/autocad_2014_pdf_dxf_reference_enu.pdf) doesn't list units, but I verified this by creating an arc in AutoCAD 2017 and saving as a DXF file and examining the output.)  `ELLIPSE` entities, on the other hand, always specify their start and end angles in radians (_page 97_).  I've added doc comments where I could to help clarify this, but more guidance could certainly be added.

Regarding communicating this to other users, I think a wiki page is a great idea.  I've enabled wiki support to the repo and I encourage you to add any useful information you find to it and I'll do the same.

nzain, 2017-04-19:
Angles... you are right, this is horrible. I don't even know what to say. I guess devs reading DXF have to manage units on their own - the responsibility should not be on this DXF package. Comments are a good start for others to build their own abstractions in whatever unit system they need.

Feet vs feet - yes, I guess this is almost impossible. I have a solution at hand, since this issue is quite common for me with all file formats.

Wiki: The wiki is a good place for me to contribute to your great package, because I can't really contribute to the code. I'll start right now. However, I'm not a native speaker - feel free to correct anything without discussion.

closed 2017-04-19 by nzain