ixmilia/dxf #69: Dxf entities mismatch

KyanWang, 2017-04-24:
Hi
First thanks a lot for you great work on this library.
I tried to import a DXF file in attach. But after imported, the entities are mismatch with the entities in DXF file.
I tried to view in LibreCAD. It works well.
Thanks for you review.
[presidio_2.zip](https://github.com/IxMilia/Dxf/files/951555/presidio_2.zip)

brettfo, 2017-04-25:
How do the entities mismatch?  E.g., what are you getting vs. what do you expect?

Using the latest sources I was able parse 104 entities, a collection of LINE and LWPOLYLINE.

KyanWang, 2017-04-25:
Hi brettfo
I am using the version 0.4.0 and the entities is 38 not 104.
[image]

brettfo, 2017-04-25:
I got 104 entities by using the latest version from source; there have been significant changes since the 0.4 NuGet package.

The README has instructions on how to build locally, but in short you need to install the [latest .NET Core SDK (check the 'Download' link)](https://github.com/dotnet/cli/releases/tag/v1.0.3) and then it's simply a matter of running `.\build-and-test.cmd` or `./build-and-test.sh` depending on your platform.

KyanWang, 2017-04-26:
Thank you brettfo. It works well now.

closed 2017-04-26 by KyanWang