ixmilia/dxf #70: Crash while reading DXF file (R14)

Happe79, 2017-05-11:
Hi,

Thank you very much for this library that I would like to use for my DXF importing. 
It is a big help.

 Now, I got a "special" DXF where it crashes while it is reading a dxf file. It crashes in "DXFLib.dll!IxMilia.Dxf.DxfAsciiReader.GetCodePair()" , with the message "Unexpected code value".

Do you may have an idea what is wrong?

File attached (you need to rename from txt to dxf).

Thanks in advance

Best regards

Happe

[822621004001.txt](https://github.com/IxMilia/Dxf/files/993556/822621004001.txt)

brettfo, 2017-05-12:
Thanks for the bug report!

The issue was that there was an extra newline at the end of the file that the parser wasn't able to handle.  You can either manually remove the final trailing CRLF sequence, or update to the latest sources; I fixed it in e42a18e1c1754f3d961f1fdddc17dac94fbae385.

closed 2017-05-12 by brettfo

Happe79, 2017-05-15:
Hello Brettfo,

Thank you very much for fast reply.

Happe