ixmilia/dxf #72: DxfPolyline causes crashes in viewer applications

brookman, 2017-08-17:
Whenever a DXF file contains a DxfPolyline our viewing software crashes. We have tried it with Autocad 2015 and Autodesk DWG TrueView 2018. We have also tried setting Is3DPolyline to true.
How to reproduce:

var dxfFile = new DxfFile();
dxfFile.Header.SetDefaults();

var vertices = new List<DxfVertex>();
vertices.Add(new DxfVertex(new DxfPoint(0, 0, 0)));
vertices.Add(new DxfVertex(new DxfPoint(1, 0, 0)));
dxfFile.Entities.Add(new DxfPolyline(vertices));

using (var fs = new FileStream("c:/tmp/file.dxf", FileMode.Create))
{
    dxfFile.Save(fs);
}

brettfo, 2017-08-17:
Thank you for the bug report!

I've fixed the issue in commit aea4eddd628895b1cecca63f5269795f2f58261b and `master` has the changes.

In short, the issue is that all DXF polylines are supposed to have a terminating `SEQEND` element after all the vertices, but that particular constructor of `DxfPolyline` was erroneously passing through `null`.

If you're not in a position to build from the latest sources (e.g., if you're using the NuGet package) then you can work around the issue by explicitly setting the `Seqend` property, either through the constructor:

``` C#
new DxfPolyline(verticies, new DxfSeqend());
//                         ^^^^^^^^^^^^^^^ this is the important part
```

or by setting the property directly:

``` C#
var poly = new DxfPolyline(vertices);
poly.Seqend = new DxfSeqend(); // this is the important part
```

brookman, 2017-08-17:
Thank you very much for the super quick response. Awesome support!

I have now generated a file with a "SEQEND" but still get TrueView to crash. Autocad gives the following error message:

Undefined group code 330 for object on line 1578.
Invalid or incomplete DXF input -- drawing discarded.

If I manually edit the file and remove all 330 group codes (330 plus the next line inside every VERTEX and the SEQEND section) the file opens fine.

brettfo, 2017-08-17:
Thanks for digging around.

I read through the spec more and the `330` code pair on an entity is only for an owner handle to a parent `DICTIONARY` object, not just any drawing item like I previously assumed so the workaround of commenting out line 209 in `DxfEntityGenerated.cs` should be fine so long as none of your entities are contained in a `DICTIONARY`, which they probably arent.

I'll keep working on a better solution and post back here when I have one.

brettfo, 2017-08-17:
I've just pushed some changes to a branch named `parent-handles`.  When you get a chance can you give that a try?

brookman, 2017-08-18:
Hello Brett
Everything works fine with the `parent-handles` branch. Thank you very much!

brettfo, 2017-08-18:
Glad it works! I've merged `parent-handles` back into `master`.