ixmilia/dxf #74: Unexpected XDATA code pair

Happe79, 2017-10-04:
Hello,

I got another file that runs into a crash. It crashes in the line 189 of DxFXDataItem

where "throw new DxfReadException("Unexpected XDATA code pair", pair);" is called.

Do you may have an idea?

Thank you

Happe

Attached file (rename it into ce117090.dxf):
[ce117090.txt](https://github.com/IxMilia/Dxf/files/1354885/ce117090.txt)

brettfo, 2017-10-05:
Thanks for the bug report!  This is an interesting problem and I'd like to get your input on it.

The issue is that while reading extended data, a point is supposed to be specified by listing its X, Y, and Z values, but in this case we have just a Y and Z value listed.  How should this be parsed?  I can see the following possibilities:

1.  Default X to 0.  **Evidence in favor of this:** elsewhere in DXF files it's not uncommon to only list X and Y values and have the Z value assumed to be 0, why shouldn't this same reasoning be applied here?  **Evidence against this:** the three values prior to the faulting line explicitly list X, Y, and Z values, all as 0.0 _and_ the Z value immediately following is also explicitly listed as 0.0; if omitting a value should default it to 0, why weren't the other 0 values omitted?  Also, what if three Y values then a Z value are specified?  Should that be read as 4 separate points, or do the last Y value and the Z value go together so that three points are the result?  How can we tell the difference?
2.  Use the last-listed X value that we've seen.  **Evidence in favor of this:** none, but it doesn't seem entirely unreasonable.  **Evidence against this:** what if the last listed X value was from a completely different entity?  What about using an X value from a different point-type (e.g., raw real triple, world point, model point, paperspace point, vector, etc.)?  There's no real easy answer.
3.  Ignore the unexpected value.  **Evidence in favor of this:** DXF readers regularly ignore code-pair values they don't understand which leads to better stability.  In fact, when new fields are added to entities, Autodesk will assign previously unused code pairs for the explicit reason that older readers will simply pass over the unexpected value instead of misinterpreting them.  **Evidence against this:** the values that were explicitly written by some other DXF library won't be preserved when read with this library.  I'd rather not throw away any data if it can be preserved, but I don't know how to preserve it when it's incomplete and doesn't make sense.

What are your thoughts?  Anything I missed?  Personally I'm leaning towards option 3 and ignoring the unexpected value simply because I'd rather omit incomplete data than assume values that might be incorrect.

Happe79, 2017-10-06:
Hello brettfo,

Thank you again very much for the fast reply.

I also think option 3 is the best way to handle this. For testing I already disabled the exception and it seems working without any problems so far.

Greetings

Happe

commit 2a996f48 (2017-10-06): ignore unexpected XDATA code pairs

Fixes #74
closed 2017-10-06 by brettfo