ixmilia/dxf #75: Maximum size of dxf-file

Idaps, 2018-01-09:
Sorry that im making an issue for this but i actually don´t know how to ask this question in an other way.

As the title says i want to know the actual size of the dxf, well not the file size but the maximum size of the element to make something like the green border in the picture.
[image]

Where and how do i get these measurements?
Because it seems that they have to be in the dxf file itself, but the values for $extmin and $extmax aren´t trustworthy.
Thanks in advance.

brettfo, 2018-01-09:
You're correct in that tools that write DXF files should ideally set the `$EXTMIN` and `$EXTMAX` variables (which this library exposes through the [DxfFile.Header.MinimumDrawingExtents](https://github.com/IxMilia/Dxf/blob/a32e8eead79b41e3be0282846e8d892071ed4ea5/src/IxMilia.Dxf/Sections/Generated/DxfHeaderGenerated.cs#L328) property), but that's not always the case.

Otherwise you'd have to compute these values yourself by iterating through all items in the [`DxfFile.Entities`](https://github.com/IxMilia/Dxf/blob/a32e8eead79b41e3be0282846e8d892071ed4ea5/src/IxMilia.Dxf/DxfFile.cs#L31) property.  The initial calculation should be pretty straight forward for things like lines and points, but can get more difficult for arcs, and there's a point of indirection for `INSERT` entities (e.g., a copy of a `BLOCK` item).  And to make this even more difficult, things like `XLINE` (construction lines that extend in both directions to infinity) don't have a proper bounding box, but the overall bounding box of the drawing should contain at least a single pixel.

I've considered adding a `.BoundingBox` property to the `DxfEntity` class that each type could override, but I simply haven't had the time yet to do so (and I'm not sure how to handle `XLINE`s).

nzain, 2018-01-10:
I could provide some GetNearestPointToLine method to calculate a pixel from
any existing bounding box of "regular" geometric items to a xline. This way
you could calculate a bounding box for regular shapes and extend to include
all xlines.

nzain, 2018-01-10:
I have methods for bounding box calculation of polylines (including arcs) ready. I could share those as well, but have to disentangle them from my code. Personally, I don't need them in the DXF context, but I'm willing to contribute if anyone is interested (scream loud to convince me).

brettfo, 2018-01-10:
@nzain, thanks for the offers!  Shortly after I responded to this issue the first time I started working on bounding box calculations for entities (including arcs), so I will shortly have that covered, but I like your idea of saving `XLINE`s until the end and expanding the bounding box by the nearest point to the existing bounding box.  Do you have (pseudo) code for a GetNearestPointToLine method handy for when I'm ready for it?

brettfo, 2018-01-10:
I've pushed my changes so far to the branch `extents`.  It doesn't yet build because I haven't computed bounding boxes for all entities yet, but it's a start.  I'll continue to fill in the remaining entities.  Note that I'll be regularly force-pushing to that branch so don't take a hard-dependency on it yet.

Once that's done I'll add tests and work on `XLINE` support.

nzain, 2018-01-11:
I thought about ways to properly compute a bounding box including xlines. Many roads lead to rome, lets first recap the problem.
- Any bounding box B is made of four line segments (the black sides).
- A xline can be axis-parallel (top red line) or skewed (other red lines).
- The xline can intersect with B (middle red line) or it is outside (others).
[image]

Lets ignore the intersection case for now (no bbox extension required). It is always sufficient to extend _one_ side (see green line) to include any xline. For non-axis-parallel lines this is always possible for _two_ sides, although one might have a shorter distance to the xline. E.g. you can extend B downwards to include the bottom red line or extend to the left. Or both. Hu!

You can now define different criteria (e.g. minimum volume, minumum line-B distance) but this might not always produce the desired result for a caller; for example, users might want to extend their (printing) area both vertically and horizontally to include a 45° xline. Thus, I'm asking the general question if it makes sense to include xlines at all into this kind of computation.

If you want to extend only one side to include a given xline, you want to look at each side of the bounding box as a line and compute line-line intersetion with the xline (drawn in green). Good explanation including performance hints is given here:
https://stackoverflow.com/a/565282/3311611

If you want to compute the closest point on the xline to a given bounding box, you do this for each edge.
Let `p=(px,py)` be the edge of `B` to test.
Let `a=(ax,ay)` be a random point on the xline and `n=(nx,ny)` a normalized (!) direction vector of the line.
The nearest point `q=(qx,qy)` on the xline is computed as `q=dotproduct(a-p,n)*n` which is
```
dp=nx*(ax-px)+ny*(ay-py);
qx=dp*nx
qy=dp*ny
```
[image]

Do this calculation for all edges to find the closest point q to the bounding box.
One of the points might be inside B, no extension required.
If not, the closest point can be used to extend the bounding box accordingly.

Basics taken from
https://en.wikipedia.org/wiki/Distance_from_a_point_to_a_line#Vector_formulation

brettfo, 2018-01-11:
@nzain I hadn't considered the possibility of expanding the bounding box by moving just one edge vs. moving a corner for the minimum distance.  Given the issues with that I'm leaning more to your suggestion of ignoring `XLINE`s altogether and treating them as non-graphical entities.

nzain, 2018-01-12:
Agreed, especially because in my last example the bounding box would only contain 1 pixel of the xline.

brettfo, 2018-01-12:
I just pushed some updates to the `extents` branch and while it builds, there's a single test failure that brings to light issues that I'd like some input on.  (I haven't added bounding box-specific tests yet.)

In short, since `DxfPoint` is a class it can be `null` which makes semi-auto-generating `GetBoundingBox()` difficult without requiring jumping through a lot of hoops.  E.g., look at the `<Extents></>` block added in `EntitySpec.xml` for the `IMAGE` entity.  If you don't want to pull it up it looks like this:

``` xml
...
<Extents>
  <Value>Location</Value>
  <Value>Location + ImageSize</Value>
</Extents>
...
```

The idea is that a method is auto-generated for you (via `EntityGenerator.cs`) that builds a bounding box containing those points, but if either `Location` or `ImageSize` are null, that causes a whole bunch of problems.

I'm not particularly fond of what I currently have and am open to suggestions, and in truth I'm leaning towards a larger refactoring of making `DxfPoint` a struct to avoid `null` issues.

brettfo, 2018-01-16:
Regarding making `DxfPoint`/`DxfVector` structs, I've created PR #78.

What are your thoughts on the (partially) breaking change?

brettfo, 2018-01-17:
I've merged the change to make `DxfPoint` and `DxfVector` structs and I've rebase the `extents` branch on top of that.

At this point all tests pass, with the caveat that not all graphical entities do the absolute correct thing with bounding boxes, but for most simple scenarios (points, lines, circles, and polylines with only straight segments) it should be correct.

brettfo, 2018-01-18:
I've merged the `extents` branch which adds the method `DxfFile.GetBoundingBox()` which should work for simple scenarios.  Saving a file also sets the `$EXTMIN`/`$EXTMAX` variables.

While the bounding box calculation isn't wholly correct, it's at least a starting point.

commit 0d3e6602 (2018-01-18): compute drawing extents on save

Fixes #75
closed 2018-01-18 by brettfo