ixmilia/dxf #79: netstandard nuget dependencies pollution

nzain, 2018-01-29:
Hi Brett,

we're moving our desktop application to .net framework 4.6.2 (compatible with netstandard2.0) and reference our own netstandard2.0 libraries now. No extra dependencies required. When I reference ixmilia.dxf.csproj directly (cloned from github) this works as expected - no extra dependencies. However, when I install the ixmilia.dxf _nuget package_, lots of netstandard nuget packages pollute my poor library. Eventually the extra dependencies don't make a difference for the final bin/release folder, but 
- the packages.config is overwhelming right now (from 2 entries up to 48) and
- relevant nuget package updates for the application are hidden between many netstandard updates (that don't make a difference anyways?).

You told me (via email last year) that this is the intended behavior for desktop apps - but since it works for direct project references, this looks wrong to me and eventually we can find a simple solution.

I've googled quite a bit, but couldn't find anything yet. There were lots of tooling issues with indirect references not flowing correctly - not my concern. I've found some threads saying it will be resolved, when netstandard2.0 is out (e.g. https://github.com/aspnet/Logging/issues/550 ) but that seems to be wrong with nuget packages. Referencing the csproj is fine, referencing the nuget package requires 46 confusing dependencies. 

Okay... I found a solution by myself: multi-targeting. Edit the csproj and change
```diff
- <TargetFramework>netstandard1.0</TargetFramework>
+ <TargetFrameworks>netstandard1.0;net45</TargetFrameworks>
```
(note that extra 's'). This will add a net45 layer to the nuget package
[image]
and consumers via nuget don't have to install netstandard dependencies. Just tested it, works as expected.

If there is no drawback (except for increased nupkg file size), it would be great to see this in the next release.

brettfo, 2018-01-30:
Seems like a totally reasonable request.  I've made the updates and pushed package version 0.7 to NuGet, but it may take some time to appear on their site.

I opted for version 0.7 instead of 0.6.1 primarily due to changing `DxfPoint` and `DxfVector` to structs, since that could break somebody.

On a related note, what version of Visual Studio are you using?  `packages.config` is an older format being phased out in favor of `<PackageReference Include="IxMilia.Dxf" Version="0.7" />` which appropriately handles transitive references and shouldn't pollute any files in your code base.

closed 2018-01-30 by brettfo

nzain, 2018-01-30:
Thanks for your quick response & action (as always + really appreciated)!

We've transitioned to VS'17 on friday. And while we've got rid of some oldschool stuff, the packages.config will stay for a while due to bugs with `PackageReference`, e.g. https://github.com/NuGet/Home/issues/4837 (install scripts not working, iirc). Unfortunately, we have quite a few heavy weight third party libs with backing native code. Install scripts modify the csproj files to copy those to the output folder. One of the better examples is sqlite.

Your nuget package is included in one particular library only, so eventually I'm able to upgrade this one. I have just tested the `PackageReference` and it correctly installs IxMilia.Dxf.**0.6**.nupkg without installing all the netstandard dependencies. The accept-license dialog popped up for all of them, but they are not showing => clean alternative. Another lesson learned. Since `0.7` is available right now, I can lean back in any case. Thanks!