ixmilia/dxf #81: Read error while opening dxf in AutoCAD 2013

nsdey, 2018-04-14:
Saved drawing from AutoCAD (2013) as DXF (test_001.dxf) , read the file in IxMilia and without any changes, save the file again from IxMilia (test_002.dxf) and then the file won't open in AutoCAD.

The error showing in AutoCAD is :
Invalid symbol table record name: on line 1570
Invalid or imcomplete DXF input -- drawing discarded.

I think there is a problem in AcDbViewTableRecord section. Please check.

[test_001.dxf.txt](https://github.com/IxMilia/Dxf/files/1910161/test_001.dxf.txt)
[test_002.dxf.txt](https://github.com/IxMilia/Dxf/files/1910162/test_002.dxf.txt)

brettfo, 2018-04-17:
A quick look seems to point to [this](https://github.com/IxMilia/Dxf/blob/f34577c0af65d2c85d2f3760fc5e5a067c88d7f0/src/IxMilia.Dxf/DxfFile.cs#L472) being the issue; `test_001.dxf` contains some plot settings objects that have an empty string for code 6 (`.PlotViewName`) which gets translated into a view without a name, but I'll have to dig in a little more to be certain.

patnewp, 2018-11-02:
@nsdey We had a similar issue and found a [workaround](https://github.com/IxMilia/Dxf/issues/94#issuecomment-435237080) that may suit your purposes.

PR #95 (2019-02-20): ensure DxfPlotSettings.PlotViewName is always useful.
Also only expose valid constructors for DxfLayout and DxfPlotSettings.

Fixes #81.
Fixes #94.
closed 2019-02-20 by brettfo