ixmilia/dxf #82: QCAD Compatibility

nzain, 2018-04-17:
I've been using this library with great success for _reading_ simple dxf files from various sources. Now, I'm trying to _write_ a simple 2D polyline like this:

```cs
DxfFile dxfFile = new DxfFile();
DxfLayer layer = new DxfLayer("testlayer", DxfColor.FromIndex(2));
dxfFile.Layers .Add(layer);
List<DxfVertex> staVertices = staticSegs
    .Select(point => new DxfVertex(new DxfPoint(point.X, point.Y, 0)))
    .ToList();
DxfPolyline staticPoly = new DxfPolyline(staVertices)
{
    Layer = layer.Name,
    Color = DxfColor.FromIndex(4),
    Thickness = 0.001,
    IsClosed = true
};
dxfFile.Entities.Add(staticPoly);
using (var fs = File.Create(@"d:\downloads\" + trackname + ".dxf"))
{
    dxfFile.Header.SetDefaults();
    dxfFile.ViewPorts.Clear();
    dxfFile.Header.Version = DxfAcadVersion.R2000;
    dxfFile.Header.DefaultDrawingUnits = DxfUnits.Meters;
    dxfFile.Header.DrawingUnits = DxfDrawingUnits.Metric;
    dxfFile.Save(fs);
}
```

Opening the resulting file in [QCAD](https://qcad.org/de/) community edition does not show the expected result: The polyline is visible, but neither layer, color, nor thickness is set. Either QCAD cannot read the format (I've tried with default AcadVersion R12, max R2018, and R2000), or the library does not write them as expected, or I'm doing something wrong. Any idea what's going wrong?

Side note: a `DxfLwPolyline` did not show up at all in QCAD.

nzain, 2018-04-17:
My collegue opened it in autocad and said it didn't work (and went home) - will add details tomorrow.

nzain, 2018-04-18:
Okay, so the DXF usage itself is even more error prone than I expected. Users without strong DXF background (like me) will run into this and scratch heads.

The evil consumer of your library _might_ expect, that some kind of automatic validation logic happens when the file is written, so that each entity validates its internal state and eventually fills missing fields or throws exceptions at the developer, who didn't know/care. Of course this is an endless route, especially because testing with one program (say AutoCad) tells little about other applications (say QCAD).

*sigh*

How do you evaluate the correctness of DXF file generation? My goal is to write code, using your library, that creates dxf files, that our customers can open with their favorite application. How can I do it right? How can I write highly compatible DXF code?

To make it more complicated, I don't have an AutoCad license (but my collegue does). Feel free to close this ticket.

brettfo, 2018-04-18:
I already attempt to [normalize](https://github.com/IxMilia/Dxf/blob/f34577c0af65d2c85d2f3760fc5e5a067c88d7f0/src/IxMilia.Dxf/DxfFile.cs#L421-L432) the DXF file on saving, but it looks like it's worth doing entity validation, too.  When I originally started this project I wanted to allow the developer to explicitly control every aspect of DXF generation, but as time has gone on I've discovered how difficult it actually is to do the right thing and I've been taking more and more control away to ensure that everything is correct, and this is a perfect example of where I should go further.  As soon as I get some free time I'll add `DxfPolyline`/`DxfVertex` normalization, but I'll have to spend some time thinking about how to handle color and thickness, since it is possible for those to differ.

You're correct in that the DXF spec is very loose and my testing methods are essentially what you've described; save a file and try to open it in QCAD or in the [Teigha Viewer] by the Open Design Alliance.  I do have some tests that directly invoke the [Teigha Converter](https://github.com/IxMilia/Dxf/blob/f34577c0af65d2c85d2f3760fc5e5a067c88d7f0/src/IxMilia.Dxf.Test/DxfCompatTests.cs#L71-L194) and some that [launch AutoCAD](https://github.com/IxMilia/Dxf/blob/f34577c0af65d2c85d2f3760fc5e5a067c88d7f0/src/IxMilia.Dxf.Test/DxfCompatTests.cs#L196-L403) to do some simple verification, but my trial license has expired.

closed 2018-04-18 by brettfo

nzain, 2018-04-25:
For your interest, I tried several minimalistic ways to store a `DxfLwPolyline`, but with no success. It never shows in QCad.

I was able to fix my layer/color issue for `DxfPolyline`: as you suggested, by explicitely specifying the polyline layer & `DxfColor.ByEntity` for each vertex.

However, I was neither able to set `DxfLineWeight.Standard` for a polyline nor for a layer. You said, each vertex has a thickness too. There is `StartingWidth` and `EndingWidth` on `DxfVertex`, but that's a `double`. The polyline additionally has `Thickness`, `DefaultStartingWidth`, `DefaultEndingWidth`, *and* `LineweightEnumValue`. How would I change the line thickness, just QCad's "standard" for the whole line? :-)