ixmilia/dxf #84: can't write file

shandoosheri, 2018-04-25:
I have been trying to simply open a file and rewrite it with no success. The file reads fine and i can view it in my viewer however, when it comes to writing it back autocad shows a blank model page but I can see the content in the layout page.

Here is the code i am using and the file I am testing.
[test.zip](https://github.com/IxMilia/Dxf/files/1945645/test.zip)

`DxfFile dxfFileReader;
                        using (FileStream fs = new FileStream(inputFile, FileMode.Open))
                        {
                            dxfFileReader = DxfFile.Load(fs);
                        }

                        DxfFile dxfFileWriter = new DxfFile();

                        foreach (DxfEntity entity in dxfFileReader.Entities)
                        {
                                dxfFileWriter.Entities.Add(entity);
                        }

                        dxfFileWriter.Header.SetDefaults();
                        dxfFileWriter.ViewPorts.Clear();
                        using (FileStream fs = new FileStream(output, FileMode.Create))
                        {
                            dxfFileWriter.Save(fs);
                        }`

kiwidave72, 2018-10-28:
We are having the same issues with reading and writing out to the same file, we are not even changing anything to the the entries all we get when opening the file in a viewer is a blank screen.

Apart from the unit tests is there a reference file that can be used to validate that the library can save and be opened in autocad. We have tried a number of different apps for viewing the output but all return a blank screen.

brettfo, 2018-11-02:
Do you have some sample drawings that you can share?  Even the act of opening and re-saving a file forces the re-computation of all item handles, so it's not a no-op like one might think.  Also, do you get a specific error message when trying to open the changed file?  It might make it easier to track down what's wrong.

closed 2020-08-31 by brettfo