ixmilia/dxf #85: Net 3.5 Support

nzain, 2018-05-16:
We use this dxf library successfully for modern WPF apps, but we have several projects for mobile applications based on compact framework (WinCE) which is supported until VS2008 and .net35. I loaded Ixmilia.Dxf source code into a net35 project and found mostly covariance issues, a missing `Guid.TryParse` call, and a linq `Zip` call. That looks promising. Since we won't get away from net35 anytime soon, I'd like to support a net35 variation of this library. I can see different routes - before I bring up a PR, I'd like to hear your oppinion.

1. I could fork and build for net35 myself. This repository stays clean and modern.
2. We could use `#if` preprocessor directives to support "the old way" in about 20 places and have a `IxMilia.Dxf.Net35.csproj` parallel to the netcore one. I can create a PR for this and I would support this one, if you're interested.
3. I can tell my boss that we have to upgrade some apps (VS and framework, too) and duplicate many csproj files (VS08+net35/VS17+net462) - probably the most time consuming option.
4. Evaluate other libraries for DXF, e.g. Teigha. Not my favorite plan.

Lets ignore 3. and 4. please :-) I know, you're working with code generation (this project doesn't even load on my machine) and eventually compiler flags are tricky? Let me know what you think.

brettfo, 2018-05-16:
I'm certainly open to option 2, particularly if we can group the `#if` blocks as much as possible, maybe even in separate files.  Is it a problem if the code generator (IxMilia.Dxf.Generator) still runs as as a `.netstandard2.0` executable?  Presumably not because all generated sources are currently part of the repository so the generator could effectively be ignored in the net35 case (once we update it's output to include `#if`).

If you've got the time to start a PR for this I'll gladly take a look and merge when it's ready.

nzain, 2018-05-16:
It will only affect the main library (or what the code generator would generate): in the netcore csproj add net35 (net45 is already there) and make the code compile for net35, too. I'll create a PR tomorrow (+16hours or so from now).

brettfo, 2018-05-24:
Fixed in #86.

closed 2018-05-24 by brettfo