ixmilia/dxf #87: DxfLwPolyline not correctly saved and loaded

lzl1918, 2018-05-21:
```CSharp
FileStream stream = File.Create("out.dxf");
DxfFile dxfFile = new DxfFile();
dxfFile.Entities.Add(new DxfLwPolyline(new List<DxfLwPolyline>()
{
    new DxfLwPolyline() {X = 0, Y = 1},
    new DxfLwPolyline() {X = 1, Y = 1},
    new DxfLwPolyline() {X = 1, Y = 0},
}));
dxfFile.Save(stream);
stream.Flush();
stream.Dispose();
```
I have codes like above. 
By setting breakpoints, there is 1 entity in `dxfFile.Entities` when `dxfFile.Save` is called.
However, when the file gets read by calling `DxfFile.Load`, there is nothing in the entity set.
I have tested that entities of type `DxfLine`, `DxfCircle`, `DxfArc`, and even `DxfPolyline` can be correctly exported, but entities of type `DxfLwPolyline` are always missing when load from file streams.

brettfo, 2018-05-21:
`LWPOLYLINE` entities were first supported in AutoCAD R14, but the [default value](https://github.com/IxMilia/Dxf/blob/821416ea712a8acad118ac01a9a3440bba94eb12/src/IxMilia.Dxf/Sections/Generated/DxfHeaderGenerated.cs#L1683) for the file version is R12.  The issue is that when saving the file, unsupported entities are simply not written.

The short answer to your problem is this:

``` C#
DxfFile dxfFile = new DxfFile();
dxfFile.Header.Version = DxfAcadVersion.R14; // <-- add this line
```

As a longer-term problem, others have suggested that `DxfFile.Save()` should throw an exception if an unsupported entity is encountered instead of [skipping it](https://github.com/IxMilia/Dxf/blob/821416ea712a8acad118ac01a9a3440bba94eb12/src/IxMilia.Dxf/Entities/DxfEntity.cs#L225).  What are your thoughts?  I'd like to hear what other people would like/expect before I make a potential change.

lzl1918, 2018-05-22:
What if unsupported entities cause exceptions in debug build, while be simply skipped in release build?

// I have no idea if nuget allows developers to release packages more than one  configurations

brettfo, 2018-05-23:
As far as I know NuGet doesn't allow the Debug/Release distinction.  I also don't want to break anybody by changing `.Save()` to throw where it didn't before.  What are your thoughts on the save method returning a collection of warnings/errors/etc.?  E.g.,

Change this:

``` C#
public void Save(Stream stream)
```

to this:

``` C#
public IEnumerable<DxfWarning> Save(Stream stream)
```

where `DxfWarning` is either a collection of `string`s (easy to write, difficult to consume) or a collection of strongly-typed messages like `UnsupportedEntity | InsufficientData | etc.` (difficult to write, easy to consume)?

That way existing users aren't getting exceptions, but they can optionally check the diagnostics if they like?

I'm also going to tag @nzain on this as he's given me great feedback in the past.

lzl1918, 2018-05-24:
Maybe in some cases, a save operation needs to be aborted or rolled back when warnings/errors occur.
Thus, in my opinion, such warnings should be accessible during save operations instead of after operations getting completed.

Personally, I have come up with an idea just now that, adding events for `DxfFile` objects. However, I don't think raising events for such light-weight operations is appropriate. And then the idea changes that override methods of `Save` can be added which pass delegates to deal warnings and/or errors.

A sample may like:
```CSharp
public enum OperationContinuation 
{
    Continue, 
    Throw,
    RollBack,
}

public delegate OperationContinuation SavingWarningHandler(IEnumerable<DxfWarning> warnings);
public delegate Task<OperationContinuation> SavingWarningAsyncHandler(IEnumerable<DxfWarning> warnings);

public void Save(Stream stream);
public void Save(Stream stream, SavingErrorHandler handler);
public Task SaveAsync(Stream stream, SavingWarningAsyncHandler handler);
```

nzain, 2018-05-24:
I guess this depends on the use case. In my scenario, I would use the very same fraction of the API (add polylines in different colors/layers) for 99% of my tasks with little/zero user interaction. Once I've learned how to `.Save()` without exceptions, this would work for all cases and customers (end-users) would not see those exceptions.

If end-users have a deeper interface into the library to possibly create invalid objects that _would_ throw on `.Save()`, they want to see localized error messages (why?) and the sophisticated `enum DxfWarning` would win here. Good enum names speak for themselves and the dev should get the idea without a long message.

Something like this

```cs
public void Save(Stream stream, Action<DxfWarning> warningCallback);
```

would give developers a lot of flexibility. I could be lazy and call like this

```cs
dxfFile.Save(stream, w => {throw new Exception(w.ToString();} );
```

or write to some log file and let the method continue. Or show a ok/cancel popup to the user, let him decide.

nzain, 2018-05-24:
Any callback could potentially throw (intentionally or not) on the consumers side. If the callback throws, this library should not have any resources (stream or similar) open - or close them via `finally` logic. The latter will leave a broken file! Better: check preconditions before writing, eventually run callbacks, and finally write the file. Probably, this is more work ;-)

brettfo, 2018-05-24:
I like the idea of `public void Save(Stream stream, Action<DxfWarning> warningCallback)`.  Since the `Save()` method just takes a stream I won't have to worry about closing any resources because I'm not the one that opened them; that'll be the responsibility of the caller to wrap in a `try`/`finally`.

For the initial pass I think I'll just write to the stream and issue the callback whenever necessary and not worry about pre-checking the file; that will get expensive and I don't think there will be much benefit.  I'll start working on this as soon as I get a chance.