ixmilia/dxf #92: Error while load dxf:“The specified conversion is invalid”

Febfire, 2018-10-10:
I install package from nuget.Everything is ok.
I just load dxf like this:
                DxfFile dxfFile;
                using (FileStream fs = new FileStream(@"E:\testifcdata\autocad_2007.dxf", FileMode.Open))
                {
                    dxfFile = DxfFile.Load(fs);
                }
I got an error:“The specified conversion is invalid”
something was wrong?

brettfo, 2018-10-10:
It sounds like there was an unexpected value in the DXF file. Can you share the file with me so I can investigate? If not, was there a file and line number reported with the exception? This won't tell me what the unexpected value was, but it could tell me where it happened so we might be able to guess what happened.

Febfire, 2018-10-12:
look forward to your reply.

Febfire, 2018-10-15:
Thanks for your reply.
got it

closed 2018-10-15 by Febfire

brettfo, 2018-10-15:
What software created the DXF file?  The indentation suggests that it was something not custom written or off the cuff.  If it's something that others are likely to use I want to make sure that my library works with it appropriately.