ixmilia/dxf #94: Read error while opening dxf - Invalid or incomplete DXF input

patnewp, 2018-10-29:
I have generated a new drawing from the acadiso.dwt template using AutoCAD v2019.  I then saved this drawing file in dxf format in AutoCAD.  Using the IxMilia.Dxf library to open the file and then save a copy appears to generate an invalid dxf file that I am unable to open in AutoCAD or BricsCAD.

AutoCAD displays the error message...

> Unknown group 102 for table on line 758.
> Invalid or incomplete DXF input -- drawing discarded.

DraftSight v2018 will open the file and includes added entities, however it displays a warning before opening the file...

> ViewTableRecord(D): Record name is empty - Ignored

```
using System.IO;
using IxMilia.Dxf;
using IxMilia.Dxf.Entities;
using Xunit;

namespace plansTest1
{
    public class UnitTest1
    {
        [Fact]
        public void ReadWriteDxf()
        {
            // Open existing drawing file
            DxfFile dxfFile = new DxfFile();
            using (FileStream fs = new FileStream(@"Blank_test.dxf", FileMode.Open))
            {
                dxfFile = DxfFile.Load(fs);
            }

            // Add a line entity to the file
            dxfFile.Entities.Add(new DxfLine(new DxfPoint(0, 0, 0), new DxfPoint(100, 100, 0)));

            // Save a copy of the file
            using (FileStream fs = new FileStream(@"Read_Write_Test.dxf", FileMode.Create))
            {
                dxfFile.Header.SetDefaults();
                dxfFile.ViewPorts.Clear();
                dxfFile.Save(fs);
            }
        }
    }
}
```

[Blank_test.dxf.txt](https://github.com/IxMilia/Dxf/files/2523551/Blank_test.dxf.txt)
[Read_Write_Test.dxf.txt](https://github.com/IxMilia/Dxf/files/2523552/Read_Write_Test.dxf.txt)

patnewp, 2018-11-02:
Hi @brettfo ,

We've got a workaround for our purposes. Your comment [here](https://github.com/IxMilia/Dxf/issues/81#issuecomment-382176060) seems to be correct.

Cheers

```
        public static DxfFile Clean(DxfFile dxfFile)
        {
            dxfFile.Header.SetDefaults();
            dxfFile.ViewPorts.Clear();
            dxfFile.Views.Clear();
            
            foreach (var dxfFileLayer in dxfFile.Layers)
            {
                dxfFileLayer.ExtensionDataGroups.Clear();
            }

            foreach (var dxfLayout in dxfFile.Objects.OfType<DxfLayout>())
            {
                dxfLayout.PlotViewName = dxfLayout.LayoutName;
            }

            return dxfFile;
        }
```

brettfo, 2018-11-02:
@patnewp Thanks for finding the workaround.  I presume that every `DxfLayout` object has a valid `LayoutName` that can be used as the default?  Do you have any `DxfPlotSettings` objects with a `null` `PlotViewName`?  `DxfLayout` inherits from `DxfPlotSettings` so I'm interested in what to do for a `DxfPlotSettings` object with an empty name that can't be populated from the layout name.

patnewp, 2018-11-02:
> I presume that every DxfLayout object has a valid LayoutName that can be used as the default?

Yes in this case.

>Do you have any DxfPlotSettings objects with a null PlotViewName?

No they are all DxfLayout with PlotViewName==""

I'm not sure what purpose the PlotViewName has. We could try just setting it to a random number and see how that goes and then you don't have any requirement that the DxfPlotSetting is a DxfLayout.

brettfo, 2018-11-07:
I just created PR #95 which I think addresses this.  Can you take a look to see if the tests cover your scenario?  Also if you have the extra time, could you pull down the changes from the `plot-view-name` branch and see if you can use your drawings?

PR #95 (2019-02-20): ensure DxfPlotSettings.PlotViewName is always useful.
Also only expose valid constructors for DxfLayout and DxfPlotSettings.

Fixes #81.
Fixes #94.
closed 2019-02-20 by brettfo