ixmilia/dxf #97: DxfColor.FromIndex(0) returns ByBlock

nzain, 2019-01-14:
I'm trying to convert RGB values to `DxfColor`. One way to do this is search the table of `DxfColor.DefaultColors` for the "closest match", e.g. by computing a squared distance.

However, the the library defines `RawValue = 0` as "ByBlock" and essentially the first color (index zero: `0xFF000000`) cannot be retreived. Is this intended? Related: #14 and #18.

Any other smart ideas, how to convert RGB to its closest DxfColor?

Thanks in advance!
  Patrick

brettfo, 2019-01-14:
Yes, this is intentional; there is no concept of `color 0` in DXF/DWG.  As for finding the closest one, I can't think of anything better than the smallest squared distance using `for (int i = 1; i < ...)`.  That being said, the color numbers in AutoCAD can be overridden, so while `color 1` is traditionally red, that's not guaranteed to be the case; the values specified in `DxfColor.DefaultColors` are simply what AutoCAD defaults to on first install.

Starting with AutoCAD 2004, entities can separately specify a 24-bit color value which I expose as the property [`Color24Bit`](https://github.com/IxMilia/Dxf/blob/5a228b514b487f787afb4bbf060d28b7f316b222/src/IxMilia.Dxf.Generator/Specs/EntitiesSpec.xml#L21), however I don't know what happens if the 24-bit color value and the index values differ.

nzain, 2019-01-15:
A 24 bit color would perfectly match my scenario (and I would not set the index at all?). Unfortunately, the `DxfLayer` doesn't have the 24 bit color property.

brettfo, 2019-01-15:
The spec only allows a color index for `LAYER`, so you'd have to set a 24-bit value on each individual entity.  As for specifying the 24-bit value vs. the index, the spec doesn't say if/when the 24-bit value is used.  The only comments are:

Regarding the color index:

> Color number (present if not BYLAYER); zero indicates the BYBLOCK (floating) color; 256 indicates BYLAYER; a negative value indicates that the layer is turned off (optional)

Regarding the 24-bit color:

> A 24-bit color value that should be dealt with in terms of bytes with values of 0 to 255. The lowest byte is the blue value, the middle byte is the green value, and the third byte is the red value. The top byte is always 0. The group code cannot be used by custom entities for their own data because the group code is reserved for AcDbEntity, class-level color data and AcDbEntity, class-level transparency data

nzain, 2019-01-16:
Thanks @brettfo for the detailed explanation. For anyone else converting custom RGB values to a best-guess color index, the following class might be useful.

```cs
    public static class DxfColorHelpers
    {
        public static DxfColor GetClosestDefaultIndexColor(byte r, byte g, byte b)
        {
            int minDist = int.MaxValue;
            int minIndex = -1;
            ReadOnlyCollection<uint> colors = DxfColor.DefaultColors;
            // index 0 is left out intentionally!
            for (int i = 1; i < colors.Count; i++)
            {
                int sqd = SquaredColorDistance(r, g, b, colors[i]);
                if (sqd == 0) // exact match
                {
                    return DxfColor.FromIndex((byte)i);
                }
                if (sqd < minDist)
                {
                    minDist = sqd;
                    minIndex = i;
                }
            }
            return DxfColor.FromIndex((byte)minIndex);
        }

        private static int SquaredColorDistance(byte r, byte g, byte b, uint otherColor)
        {
            (byte r2, byte g2, byte b2) = ToRgb(otherColor);
            return (r - r2) * (r - r2)
                 + (g - g2) * (g - g2)
                 + (b - b2) * (b - b2);
        }

        public static (byte, byte, byte) ToRgb(uint color)
        {
            //byte a = (byte)(color >> 24);
            byte r = (byte)(color >> 16);
            byte g = (byte)(color >> 8);
            byte b = (byte)(color >> 0);
            return (r, g, b);
        }

        public static uint FromRgb(byte r, byte g, byte b)
        {
            const byte a = 0xFF; // alpha not used
            return (uint)(a << 24 | r << 16 | g << 8 | b << 0);
        }

        public static string ToHexString(this DxfColor color)
        {
            if (color.IsIndex)
            {
                uint argb = DxfColor.DefaultColors[color.Index];
                return ToHexString(argb);
            }
            return color.ToString();
        }

        public static string ToHexString(uint color)
        {
            return "0x" + color.ToString("X8");
        }
    }
```

closed 2019-01-16 by nzain

brettfo, 2019-01-16:
I'd take your helper functions as a PR if you think they'll be useful to you.

nzain, 2019-01-16:
Not worth the effort for now - I'm experimenting with an export into a 3D dxf file for our customers... but this is really troublesome, especially because not all cad apps use / understand the same properties. If only a few more problems come up, this project is dead.