ixmilia/dxf #102: DxfBoundingBox does not account for Bulge

kmckee, 2019-02-16:
Hello,

I believe I found a bug where the BoundingBox is not taking into account the `Bulge` of PolyLines.  I've attached a sample dxf and some sample code below that demonstrates what I'm seeing.

Running the following code:

```csharp
var readStream = File.OpenRead(attachedFilePath);
var document = DxfFile.Load(stream);
var box = document.GetBoundingBox();
Console.WriteLine($"Maximum X is reported as: {box.MaximumPoint.X} but it should be 0.5");
Console.WriteLine($"Minimum X is reported as: {box.MinimumPoint.X} but it should be -2.0");
```

Outputs the following:

```
Maximum X is reported as 0.00 but it should be 0.5
Minimum X is reported as -1.50 but it should be -2.0
```

The DXF file is a PolyLine with 4 points oriented like so:

```
    b---------------------a
    |                     |
    c---------------------d
    ^ min x               ^ max x
```

Except that points `b` and `d` have their `Bulge` set to 1.0, so it should be a slot shape like this:

```
    b---------------------a
|                             |
    c---------------------d
^ min x                       ^ max x
```

The BoundingBox does not account for the Bulge values.

kmckee, 2019-02-16:
Here's the file.  You'll have to switch the file extension back to DXF, GitHub wouldn't allow it otherwise.
[Slot_1x1.5_Inches.txt](https://github.com/IxMilia/Dxf/files/2872445/Slot_1x1.5_Inches.txt)

nzain, 2019-02-17:
As far as I remember, I have some code for exactly this. Will report on
Monday.

nzain, 2019-02-18:
@brettfo I reproduced the issue with a unit test and my proposed solution is creating a `DxfArc` from a `DxfVertex` with bulge (and its successor), since `DxfArc` correctly calculates its bounding box. It will take a while to properly implement this conversion, but I'll get back with a PR soon.

brettfo, 2019-02-20:
Fixed via #104.

closed 2019-02-20 by brettfo