ixmilia/dxf #103: DxfArc.GetBoundingBox incorrect for StartAngle>EndAngle

nzain, 2019-02-18:
Consider a counter-clockwise arc with startangle=90° (top point) and endangle=0° (right point) centered at `(0,0)`. It defines a 270° sweep and looks like this:

[image]

The current bounding box calculation calls into `DxfEntityExtensions.ContainsAngle` which reverts the order of the angles and assume an arc from 0° to 90°. The resulting bbox `min=(0,0) max=(1,1)` is wrong.

Correct handling of angles is always tricky, mostly because of the 360° boundary (think of `start=359°, end=1°` and other boundary crossing cases). Unfortunately, this blocks a relatively simple solution for #102.

nzain, 2019-02-19:
Since `DxfArc` seems to be CCW always, the correct way to test if an angle is contained in the arc includes three cases:

If `startAngle < endAngle` then target angle must live between start and end.

Otherwise the CCW arc end goes beyond 360° and we split the arc into two parts: "from start to 360°" and "from 0° to end". Those can be easily tested with normalized angles.

brettfo, 2019-02-20:
Fixed via #104.

closed 2019-02-20 by brettfo