ixmilia/dxf #105: Arcs outside

Happe79, 2019-05-20:
Hello,

I got another issue with an older Autocad version (Release 9). Some arcs of the dxf file are transformed. I attached the file (rename .txt to .dxf). I guess it has something to do with the group values 210, 220 and 230.

Can you may take a look. Thx

Happe
[test.txt](https://github.com/ixmilia/dxf/files/3198148/test.txt)

brettfo, 2019-05-20:
You're right that a bunch of the arcs have a normal of `(0, 0, -1)` which causes them to look like they're backwards.  The attached file was saved as an R14 drawing (`AC1014` on line 8).  Am I right in assuming that the file you attached was originally an R9 drawing that you opened and re-saved as R14 with this library?  Do you still have access to the original R9 drawing that you can send?  I'd like to step through the arcs to see what caused the negative normal (codes 210-230 like you said.)

Happe79, 2019-05-21:
No, it should be the original one. But you are right it is Release 14. For some stupid reason I read AC1004.

brettfo, 2019-05-21:
What created the original .dxf file?  You'll likely have to correct the backwards normal in that.  Otherwise you could use this library to open the drawing and manually correct the normals (assuming the start/end angles are correct):

``` C#
// using IxMilia.Dxf;
// using IxMilia.Dxf.Entities;
DxfFile file = DxfFile.Load(@"C:\path\to\file.dxf");
foreach (DxfEntity entity in file.Entities)
{
    if (entity is DxfArc)
    {
        DxfArc arc = (DxfArc)entity;
        arc.Normal = DxfVector.ZAxis; // this forces all normals to be (0, 0, 1).  You may want to double-check that it's (0, 0, -1) to begin with
    }
}

file.Save(@"C:\path\to\file-with-fixed-normals.dxf"); // write to a different location so you don't loose the original file
```

Happe79, 2019-05-22:
The dxf file are created from "Proliner", a system to measure kitchens.

The code does not solve the problem but it was the clue to take a deeper look into my implementation and I found out that I do not pay attension to normals. 
How do I need to deal with the nomal values? In this case I know I have to mirror the arcs on the Y axi (then the arcs on the correct position with the right orientation). But is there a general way of how to transform the arcs if the normal is not (0, 0, 1)? Till now I just used the center point, the radius, start and end angle and I was able to convert the dxf file into my own format.

I attached an image to show you how it looks like when I make my dxf import. 

[image]

brettfo, 2019-05-23:
I went digging through the DXF spec and found the [Arbitrary Axis Algorithm](http://help.autodesk.com/view/OARX/2020/ENU/?guid=GUID-E19E5B42-0CC7-4EBA-B29F-5E1D595149EE) which uses the normal vector to compute an object/entity-specific right X-axis.  I fixed my implementation of that computation in commit 1e4d2af780fc81982b1085bf9468d72c3ff6dd76.

You should be able to use the arc's normal and x-axis (as computed in [`DxfVector.RightVectorFromNormal()`](https://github.com/ixmilia/dxf/blob/1e4d2af780fc81982b1085bf9468d72c3ff6dd76/src/IxMilia.Dxf/DxfVector.cs#L142)) to then compute the start/end points of the arc and convert that to your own format.  I'll try to revisit this issue to add helper methods to do just that, but I currently don't have the bandwidth.

Thank you for finding and reporting the issue, let me know what else I can help with.

closed 2019-05-23 by brettfo

Happe79, 2019-05-28:
Sorry for the late reply. Now I could solve the problem. THX