ixmilia/dxf #106: Can you check this file?

HueJonghyukParkClothingTechLLC, 2019-07-30:
[reexport80973-1.zip](https://github.com/ixmilia/dxf/files/3449059/reexport80973-1.zip)

Hi,
I'm working on my project with ixmilia.dxf library but I can not open this file with this library.
Can you check this file works in this library?

if you need more information for fix the problem. you can contact me by email.
hue.park@clothingtechllc.com

Thanks

brettfo, 2019-07-30:
How did you get the DXF file?  According to the `$ACADVER` variable, this was produced as an AutoCAD R10 file (`AC1006`).

Values in a DXF file come in pairs; a number that specifies what the type of the value is, and on the next line the actual value.  For example, lines 145/146 in the file specify code `250` with a value of `2`, but according to the [latest spec](http://help.autodesk.com/view/OARX/2020/ENU/?guid=GUID-2553CF98-44F6-4828-82DD-FE3BC7448113), code 250 isn't supported and that's where it [stops reading](https://github.com/ixmilia/dxf/blob/1e4d2af780fc81982b1085bf9468d72c3ff6dd76/src/IxMilia.Dxf/DxfCodePair.ExpectedType.cs#L93).

Otherwise replacing all 63 occurences of "`  250`" with "`  50`" (a valid, but meaninless code at this point) allows the file to be opened.

HueJonghyukParkClothingTechLLC, 2019-07-30:
Thank for your quick responding!

I got this  file from clo 5.0 software
URL: https://www.clo3d.com/obt/5_0

This is very famous garment design tool.
So do you have any plan for update your library or should I just manually fix and use ixmilia library?

Thanks,
Hue

Hue Jonghyuk Park
Clothing Tech LLC
Software Developer
Email: hue.park@clothingtechllc.com<mailto:hue.park@clothingtechllc.com>
Office: 5113 Southwest Parkway, Suit 240, TX 78735
Mobile: +1-619-715-2036

brettfo, 2019-07-30:
I've contacted the CLO company to try to figure out the intended use of the `250` code/value pair, so I'll wait to officially update this library until I hear back from them.

In the meantime you can make a local change to effectively ignore this `250` value by adding two lines to the `src/IxMilia.Dxf/DxfCodePair.ExpectedType.cs` file like this:

``` diff
             else if (code == 1071)
                 expected = typeof(int);
+            else if (code == 250)
+                expected = typeof(string);
             else
                 throw new ArgumentOutOfRangeException("code", "Unknown type code " + code);
```

closed 2019-08-01 by brettfo

brettfo, 2019-08-06:
I just heard back from the CLO developers and the only non-standard codes they use in DXF files are the 250 values in POLYLINEs, so with the merge of #111 this library should be good to go regarding CLO-produced files.

PR #111 (2019-08-01): Support clo
Fixes #106.