ixmilia/dxf #107: Problem with special characters in layer name (and probably any other text)

vemo-france, 2019-07-31:
Hi,

I'm having some problems with encoding/codepage and special characters.

For instance, in [this file](https://raw.githubusercontent.com/vemo-france/public_stuff/master/ixmilia/PL14.dxf), there is a layer named "Repère pièce", but it appears as "Rep?re pi?ce".

I found that pair in the file:

```
9
$DWGCODEPAGE
3
ANSI_1252
```
ANSI_1251/Windows-1252 is the ["latin" codepage](https://en.wikipedia.org/wiki/Windows-1252).

Are codepages supported in IxMilia.Dxf ?

By the way, your library will be used to process a few thousands dxf files a day in the web frontend of a metal laser cutting shop. Thanks a lot for your work !

brettfo, 2019-08-01:
It's not a code-page issue, per se, but rather an issue with knowing when the file might be ASCII or UTF-8 encoded.  I've created #109 which I think will address your issue, can you give that a try?  It's the branch `unicode`.

vemo-france, 2019-08-01:
It works flawlessly with the sample files I got.

Still, if the file is ASCII (or rather [Extended ASCII](https://en.wikipedia.org/wiki/Extended_ASCII)) encoded , how do you interpret codes from 0x80 to 0xFF without using codepages ?

I guess it works in my case because ANSI_1252 and unicode share the same characters from 0x00 to 0xFF, but I'm not sure it'll work with, say, an ASCII encoded DXF file with a turkish codepage. Will it ? Can't find one to do the test !

Anyway, it solves my problem. Thank a lot for adressing that issue so fast !

brettfo, 2019-08-01:
The [official spec](http://help.autodesk.com/view/OARX/2020/ENU/?guid=GUID-2553CF98-44F6-4828-82DD-FE3BC7448113) (scroll down to "Storage of String Values") states that the format only supports ASCII (which I interpret to mean 0-127) with the upper 128 values being encoded in the format `\U+xxxx` where `xxxx` is a hex unicode value.  For an example look at the tests I added in PR #109.

nzain, 2019-08-01:
Often ASCII and ANSI (ASCII+WinCodepage) are confused. I think C# will use
the current Codepage (whichever that may be) to decode bytes > 127 when you
use `Encoding.ASCII`.

brettfo, 2019-08-01:
@nzain I'm certain that I don't understand the ASCII vs. ANSI divide.  Is there a scenario where different values in the `$DWGCODEPAGE` header variable would (or should?) make a difference in how values 128-255 are interpreted?  E.g., is this something that this library should handle, or is it handled by the host OS/runtime?

closed 2019-08-01 by brettfo

nzain, 2019-08-01:
You write ASCII (lower 7 bits, bytes 0-127) when you want the result be portable across languages (spoken and programming as well) because those lower 7 bits are the same for ASCII, ANSI (regardless of codepage), and even UTF8.

Whenever you read text you have to know the encoding, because it is almost never obvious. C# assumes UTF8 (which works with ASCII values 1-127 just fine) unless you specify something else. When you use `Encoding.ASCII` I _think_ it will use the current codepage. And if the file was stored in Japan, it might specify a japanese code page in the DXF header (defining a particular meaing for bytes 128-255) that _could_ be loaded in France to display japanese characters on a french computer. This is uncommon, for sure, and I don't know how you load a code page in C# explicitely.

Since UTF8 is the new gold standard, even for DXF since 2007, just using `Encoding.ASCII` (which probably uses the default code page) should handle all cases. Lets hope we never hear about this again :)

PR #109 (2019-08-01): read/write unicode string values
According to the spec it's unknown if the file can be read as UTF-8 until the version in the header has been encountered, hence the switch to the UTF-8 reader if `$ACADVER >= AC1021`.  If that switch wasn't made, UTF-8 characters are specified with the well-known `\U+xxxx` sequence.

Fixes #107.
Fixes #108.