ixmilia/dxf #108: Cannot read MTEXT with unicode characters

nimishmistry, 2019-07-31:
Cannot read MTEXT with unicode characters e.g. "äöüÄÖÜß" or "你好".

Noticed that there is only a "DxfAsciiReader", which supports reading only ASCII data

```C#
var streamReader = new StreamReader(stream, Encoding.GetEncoding("us-ascii"));
```

Any particular reason for not supporting unicode?

Is there a possibility to add such a support?

brettfo, 2019-08-01:
According to [the spec](http://help.autodesk.com/view/OARX/2020/ENU/?guid=GUID-2553CF98-44F6-4828-82DD-FE3BC7448113) (scroll down to "Storage of String Values") the file's encoding isn't known until the `$ACADVER` header variable has been read.  I created PR #109 which properly switches to UTF-8 mode, can you try that and see if it fixes your issue?  Branch `unicode`.

nimishmistry, 2019-08-01:
Did some basic tests and it seems to fix the issue.

closed 2019-08-01 by brettfo

nimishmistry, 2019-08-06:
When will we have a new version?

brettfo, 2019-08-06:
I don't have a plan to release a new version of the package in the short term, so if you need to consume the recent changes you can either clone/build/reference this locally or you can `dotnet pack` to a folder and add that as a package source in your local `NuGet.config` or modify the MSBuild variable `$(RestoreSources)`.

nimish-trumpf, 2019-09-11:
I'm taking a different approach for using the package. I have a local nuget repository, to which I plan to push the package. However, I'm a bit concerned about the versioning. I plan to version my packages as 0.72.x, this will ensure that it does not conflict with the versions deployed by you. I believe, I just have to update the "version.txt" for updating the version numbers?

brettfo, 2019-09-11:
Correct, to produce a new version you can:

1. Update `version.txt`.
2. `build-and-test.cmd`/`build-and-test.sh`.
3. `dotnet pack --no-restore --no-build -c Release -o "C:\path\to\local\repository" "src\IxMilia.Dxf\IxMilia.Dxf.csproj"` (copied from [here](https://github.com/ixmilia/dxf/blob/ac4fc8967bf26ef34f172e12e78ffc38949f981f/azure-pipelines.yml#L32)).

PR #109 (2019-08-01): read/write unicode string values
According to the spec it's unknown if the file can be read as UTF-8 until the version in the header has been encountered, hence the switch to the UTF-8 reader if `$ACADVER >= AC1021`.  If that switch wasn't made, UTF-8 characters are specified with the well-known `\U+xxxx` sequence.

Fixes #107.
Fixes #108.