ixmilia/dxf #110: Files that make DxfFile.Load throw

vemo-france, 2019-08-01:
Hi,

I've tested IxMilia on 53 000+ DXF files. 180 of those (some of which may be invalid files) make DxfFile.Load throw.

Here is a summary of exception messages:

```
+-------------------------------------------------------------------------------------+------------+
| Exception message                                                                   | Quantity   |
+-------------------------------------------------------------------------------------+------------+
| Unknown type code 83. Parameter name: code                                          |        111 |
| Unexpected code value                                                               |         63 |
| Unsupported value ' 2,5 ' for code '40'                                             |          2 |
| Unsupported value ' 0,2 ' for code '40'                                             |          1 |
| Index and length must refer to a location within the string. Parameter name: length |          1 |
| Expected value                                                                      |          1 |
| Object reference not set to an instance of an object.                               |          1 |
+-------------------------------------------------------------------------------------+------------+
```

I can send files to interested contributors (40mo zip).

brettfo, 2019-08-01:
I'd love to have a copy of those if you can share them.  My personal email is `brett.forsgren` `@` `outlook.com`.  If they're too big to email I can create a folder on OneDrive and give you access if you let me know your email address.

brettfo, 2019-08-02:
I just received the files, thank you.  I'll start working my way through the issues and add updates here.

brettfo, 2019-08-02:
I've done the initial analysis of the files you sent and here's the breakdown.  (Apologies to those who don't have access to the files, but @vemo-france has requested that I keep them private.)

| Exception message | Quantity | Analysis | Next steps |
|-|-|-|-|
| Unknown type code 83. Parameter name: code | 111 | Post R13 binary DXF file with different encoding. | Filed #112.  I have a good idea of how to approach this. |
| Unexpected code value | 63 | Not DXF files, actually DWG. | Possibly detect something far out of the ordinary and throw a better exception?  Still pondering this one. |
| Unsupported value ' 2,5 ' for code '40' | 2 | Improperly written text style height evidenced by the proper value '2.5' being written for code 42 a few lines below. | No action for this repo.  The library that wrote that file should use invariant culture when writing decimal values. |
| Unsupported value ' 0,2 ' for code '40' | 1 | Improperly written text style height evidenced by the proper value '.2' being written for code 42 a few lines below. | No action for this repo.  The library that wrote that file should use invariant culture when writing decimal values. |
| Index and length must refer to a location within the string. Parameter name: length | 1 | Improperly truncated unicode sequence on line 6262. | No action. |
| Expected value | 1 | File is truncated/not complete. | No action. |
| Object reference not set to an instance of an object. | 1 | File is empty. | Fixed in commit abddd21b3e3d783cae61f25d7f16bbd8b0774c60. |

closed 2019-08-02 by brettfo