ixmilia/dxf #116: Show DxfEllipse in canvas

sinaone1, 2019-08-23:
Hello
I can convert DxfEllipse to EllipseGeometry to show it in canvas when it doesn't have angle.
But if there is angle for my DxfEllipse I don't know how can I show it.
Is there any example for show DxfEllipse  in canvas?
Thank you.

brettfo, 2019-08-23:
Ellipse transformations can be tricky.

For the purposes of this question I'm going to assume that the ellipse's normal is along the z-axis, `(0, 0, 1)` and that you're talking about rotation around that axis; ellipse transformations **not** around the z-axis are beyond my ability, but others have found help in [this](https://www.geometrictools.com/Documentation/PerspectiveProjectionEllipse.pdf) document.  Z-axis rotations are represented in DXF ellipses by the `MajorAxis` property which can be used to determine the angle:

``` C#
var rotationAngleInRadians = Math.Atan2(ellipse.MajorAxis.Y, ellipse.MajorAxis.X);
var rotationAngleInDegrees = rotationAngleInRadians * 180.0 / Math.PI;
```

Assuming you're using `EllipseGeometry` from WPF, you can use the [`Transform`](https://docs.microsoft.com/en-us/dotnet/api/system.windows.media.geometry.transform?view=netframework-4.8#System_Windows_Media_Geometry_Transform) property to apply a rotation based on the values computed above:

XAML:
``` xml
<!-- N.b., this XAML snippet won't work exactly as I've typed it, but the idea is the same. -->
<EllipseGeometry Center="{Binding Path=Center}" RadiusX="{Binding Path=MajorAxis.Length}" RadiusY="{Binding Path=MinorAxis.Length}">
  <EllipseGeometry.Transform>
    <RotateTransform CenterX="{Binding Path=Center.X}" CenterY="{Binding Path=Center.Y}" Angle="{Binding Path=RotationAngleInDegrees}" />
  </EllipseGeometry.Transform>
</EllipseGeometry>
```

C#:
``` C#
dxfEllipse = ...; // the DxfEllipse class defined in this library
var ellipseGeometry = new EllipseGeometry(new Point(dxfEllipse.Center.X, dxfEllipse.Center.Y), dxfEllipse.MajorAxis.Length, dxfEllipse.MajorAxis.Length * dxfEllipse.MinorAxisRatio);
ellipseGeometry.RotateTransform(rotationAngleInDegrees, dxfEllipse.Center.X, dxfEllipse.Center.Y); // rotation angle as computed above
```

sinaone1, 2019-08-24:
Perfect
Thank you it's working now.

closed 2019-08-24 by brettfo