ixmilia/dxf #129: Nuget update for .net core 3.1.

ramubhai, 2020-02-12:
Now that we have an LTS version (3.1) of dot net core and nuget version seems to be almost 1.5 years old, may I request nuget update of this package?

Further, https://github.com/ixmilia/dxf/blob/master/src/IxMilia.Dxf/IxMilia.Dxf.csproj does not include netstandard3.1 while the main README recommeds to install .netcore 3.0 in "Building Locally" section.

nzain, 2020-02-12:
What would be the benefit? The application defines the runtime, the library target has no impact on this (thus, this is not about performance).

A library should target a low netstandard for maximum compatibility. A `netcoreapp3.1` can reference a `netstandard2.0` library, but an app2.0 cannot reference a netstd3.1 library. Yes, there is multi targeting, of course.

There is some benefit to reduce dependencies, but I'm not sure if this would be the case for 2.0 -> 3.1.

Finally, the library author may be interested to use the latest stuff, which might only be available on the newest netstandard. I don't think this is your point :) Please enlighten us/me - thanks!

nzain, 2020-02-12:
Oh, the nuget update is of course very useful!

ramubhai, 2020-02-13:
I guess I was too quick to post this issue. More so because of my newbie know how of .net. 

I refereed the nuget from my project targeting netcore3.1 desktop application and was pleasantly surprised to see it compile without any warning. I need to target netcore3.0+ for desktop application support.

Side Note: I am having difficulty finding sample code using his library. Would the library authors accept pull request adding new samples?

brettfo, 2020-02-13:
@ramubhai PRs for samples would be very much appreciated.

I'll update the README regarding 3.0/3.1 and work on publishing an update to the official NuGet package soon.

brettfo, 2020-02-13:
Just pushed version [0.7.3](https://www.nuget.org/packages/IxMilia.Dxf/0.7.3) to NuGet.

closed 2020-02-13 by brettfo