ixmilia/dxf #130: Migrate to MIT license

brettfo, 2020-04-17:
I'd like to migrate this repo to be under the MIT license, but first I need signoff from all past contributors.  If each of you tagged below agree to this, could you add a comment below indicating this?

@nzain relevant commits:
- Support and build for net35 (6472003b3306fd2719c96bbcf261dc3fda8fcc49)
- Set default layer for DxfBlock (a82e53b7555a3b7d9421d6c654e336b2238a0689)
- fix DxfArc bounding box calculation and consider bulge when calculati… (2e8dacd2717cbae918a62eed5f491fcc2ec54288)
- Improve Autodesk Compatibility (#122)
- Write missing subclass marker for DxfDictionaryWithDefault (#127)
- Assign handles to Block.Entities to improve Autodesk compatibility (#126)
- Dimension entities should only write BlockName if set explicitely (#125)

@lgrzywac relevant commits:
- Fix reading point triple real (#71)

@ZoiSJP relevant commits:
- Replaced the Property Weight by a property Weights which includes a w… (2e5995e0de7ed761eaf768e0bdddd6c976eecdbd)

nzain, 2020-04-18:
Agreed, but could you explain why? Consumers may not notice the change and produce a bad EULA or similar.

Personally, I prefer the MIT license due to its brevity. If I understand both (most certainly not)... this change should not block any consumer of this library. Please correct me, if I'm wrong!

brettfo, 2020-04-20:
I was recently contacted by an individual that wanted to use this library in a closed-source product and from my understanding of their situation as well as the licensing agreements in general (not a lawyer, though), consumption of MIT-licensed components is easier, in part due to its brevity, like you said.

Since the original purpose of this project was to be useful to others, I feel this goes along that path of making it easier to consume.

brettfo, 2020-05-11:
Quick ping to @lgrzywac and @ZoiSJP.  I'd like to migrate the license of this project to MIT and my understanding is that I need the approval of all past contributors.  If you could reply to this issue with your consent (or not), then I'll have a record of it.

lgrzywac, 2020-05-31:
@brettfo You have my approval

brettfo, 2020-06-01:
I've attempted to contact @ZoiSJP to re-license their contribution, but I've received no reply.  That user's only GitHub activity is the commit listed above and directly emailing the address on the commit got bounced back as invalid.

Given the above I'm going to consider that account, and subsequently the code contribution, as abandoned and move forward with the MIT re-license.

closed 2020-06-01 by brettfo