ixmilia/dxf #132: Can't save DxfEllipse

MuziburRahman, 2020-06-06:
What's the parameters in the constructor of the DxfEllipse stands for?
The first parameter is understandable: center of the ellipse. But what about the others?
I have tried this:

`dxf_file.Entities.Add(new DxfEllipse(new DxfPoint(2, 3, 0), new DxfVector(100, 100, 0), 0.5));`

it doesn't show any ellipse in eDrawing.
I have also tried specifying StartAngle And EndAngle with no luck.

What does the vector (2nd parameter) do, and what does the 3rd parameter do?

brettfo, 2020-06-06:
Ellipse not saving: the DXF spec only added ellipses in R13 so they're only written if the drawing version is at least that. Set the `DxfFile.Header.Version` property to at least `DxfAcadVersion.R13`. AutoCAD is really picky about what it'll read so it defaults to writing R12 because that's the most widely accepted version, so be sure to check that files written appear properly in any other programs that you may want to interop with.

About the constructor parameters: the names of the constructor parameters should appear in whatever you're using to code, e.g., Visual Studio. The second parameter is the major axis, which is a vector from the center of the ellipse to the farthest point. The length of that vector is like the large radius. The third parameter is the minor axis ratio, which should be between 0 and 1 and specifies how long the small radius is compared to the large. The start and end parameter values are used if you don't want the ellipse to be fully closed, but instead an elliptical arc. Be warned though, the spec says that regular circular arcs specify start/end angles in degrees, but elliptical arcs use radians.

MuziburRahman, 2020-06-07:
Yes, that solved it. I am closing the issue. 
But it'd be nice to have this header-issue in the summary of the constructor.

closed 2020-06-07 by MuziburRahman