ixmilia/dxf #133: Generated dxf files do not open in Autocad 2015

uzername, 2020-06-25:
Hello. I am generating dxf files with ixmilia. But generated files do not open in autocad 2015. They open fine in qcad, where they can be saved and then they are opened. I have compared files before and after opening in QCAD. I assume that this happens because AcDbSymbolTable entry is not added to file by ixmilia. QCAD adds it. Here is a sample code:

```
        private void PerformDxfLayerRemoving(String in_inpFilePath, String in_outFilePath, String in_layerToRemove)
        {
            DxfFile dxfFile;
            DxfFile dxfFileOut = new DxfFile();
            using (FileStream fs = new FileStream(in_inpFilePath, FileMode.Open))
            {
                dxfFile = DxfFile.Load(fs);
            }
            foreach (DxfLayer dxfLayeritem in dxfFile.Layers)
            {
                if (dxfLayeritem.Name != in_layerToRemove)
                {
                    dxfFileOut.Layers.Add(dxfLayeritem);
                }
            }
            foreach (DxfLineType dxfLineTypeitem in dxfFile.LineTypes)
            {
                dxfFileOut.LineTypes.Add(dxfLineTypeitem);
            }
            foreach (DxfEntity entity in dxfFile.Entities)
            {
                if (entity.Layer != in_layerToRemove)
                {
                    dxfFileOut.Entities.Add(entity);
                }
            }
            using (FileStream fs2 = new FileStream(in_outFilePath, FileMode.Create))
            {
                dxfFileOut.Save(fs2);
            }
        }
```
It reads input file and saves all Entities from it ignoring those which belong to certain layer

uzername, 2020-06-25:
I am getting error, "unknown group 74"

uzername, 2020-06-25:
[dxftesting.zip](https://github.com/ixmilia/dxf/files/4830949/dxftesting.zip)
here are two example dxf files. they don't open in autocad, but work in qcad

uzername, 2020-06-25:
For NEWsn_arc3.dxf error message is reported by autocad:
`Error in the VIEW table Invalid character record name: on line 3464.`
For another file error is:
`Invalid record name in character table: on line 2996.`

brettfo, 2020-07-08:
Thanks for filing an issue.

It appears that the problems with the files ultimately come from [here](https://github.com/ixmilia/dxf/blob/61c220453264b972ad2c89ea3f96f717fb2fce93/src/IxMilia.Dxf/DxfFile.cs#L493) where the `VIEW` table is populated with items from the `PLOTSETTINGS` objects, but the `.Name` property isn't getting set.  I'll try to figure out what the restrictions/requirements are around the values for that field.  In the meantime can you try to delete all `PLOTSETTINGS` objects before saving (and therefore ensuring bad `VIEW` table items aren't created)?  You should be able to do so with this:

``` csharp
DxfFile dxfFile = ...;
for (int i = dxfFile.Objects.Count - 1; i >= 0; i--)
{
    if (dxfFile.Objects[i] is DxfPlotSettings)
    {
        dxfFile.Objects.RemoveAt(i);
    }
}
```

You may also want to check that all existing `VIEW`s have a name set:

``` csharp
if (dxfFile.Views.Any(v => string.IsNullOrEmpty(v.Name))
{
    throw new Exception("found a view with an empty name");
}
```

---

On a related note, you'll have much better luck modifying the same `DxfFile` object in memory, rather than copying items over; there are a ton of relationships between items in a file and it's (obviously) difficult to catch everything.

You can delete a layer like so, which is very similar to what you have above:

``` csharp
public void RemoveLayerFromDrawing(string inFile, string outFile, string layerToRemove)
{
    var dxfFile = DxfFile.Load(inFile);
    for (int i = dxfFile.Layers.Count - 1; i >= 0; i--)
    {
        if (dxfFile.Layers[i].Name == layerToRemove)
        {
            dxfFile.Layers.RemoveAt(i);
        }
    }

    for (int i = dxfFile.Entities.Count - 1; i >= 0; i--)
    {
        if (dxfFile.Entities[i].Layer == layerToRemove)
        {
            dxfFile.Entities.RemoveAt(i);
        }
    }

    dxfFile.Save(outFile);
}
```

uzername, 2020-07-09:
Thank you for comment!

In my project I had to write another C++ module using dxflib from qcad to mitigate issue. 😂 

It would be great to have the above mentioned DxfPlotSettings fix in core, so bad empty views are not written to file.

brettfo, 2020-07-18:
I've experimented a bit and found that AutoCAD doesn't create `VIEW` table entries for `PLOTSETTINGS` objects, so I removed that requirement in commit e374c5a3303c3113e59b57080ea20714bf9b136f.  It still might be possible to open a file with invalid settings (e.g., an empty `Name` property on a `VIEW`), but no new view will be created with an empty value.

I've also added commit 2adb1d54d6c7beb8dcf49f18b6e452321e021e6a that ensures a user-created table item (e.g., a `VIEW`) has a non-empty name.

closed 2020-07-18 by brettfo