ixmilia/dxf #134: Index out of bound exception

MuziburRahman, 2020-06-25:
I am using the library in my app.
I have received a recent crash report that was caused by a IndexOutOfBoundException inside the library.

Here's the relevant information and stacktrace:

Windows.Desktop | 10.0.18363.900 | X64 | LENOVO | 80QR
Index was out of range. Must be non-negative and less than the size of the collection. Parameter name: index
at System.Collections.Generic.List1.get_Item(Int32) + 0x38 at IxMilia.Dxf.Collections.ListWithPredicates1.get_Item(Int32) + 0x10
at IxMilia.Dxf.Entities.DxfLeader.PostParse() + 0x72
at IxMilia.Dxf.Entities.DxfEntity.PopulateFromBuffer(DxfCodePairBufferReader) + 0xcc
at IxMilia.Dxf.Entities.DxfEntity.FromBuffer(DxfCodePairBufferReader) + 0xa1b
at IxMilia.Dxf.Sections.DxfEntitiesSection.EntitiesSectionFromBuffer(DxfCodePairBufferReader) + 0xe1
at IxMilia.Dxf.Sections.DxfSection.FromBuffer(DxfCodePairBufferReader, DxfAcadVersion) + 0x165
at IxMilia.Dxf.DxfFile.LoadFromReader(IDxfCodePairReader) + 0xb8
at IxMilia.Dxf.DxfFile.Load(Stream, Encoding) + 0x92

brettfo, 2020-06-25:
Thanks for forwarding the issue!  It's been fixed in the latest in `main`.

closed 2020-06-25 by brettfo

MuziburRahman, 2020-06-26:
Will the nuget package be updated?

brettfo, 2020-06-26:
Just pushed `main` as version 0.7.4 to NuGet.

commit 3dc6f120 (2020-06-25): properly ready incomplete leader points

Fixes #134.