ixmilia/dxf #135: GetBoundingBox method for DxfSpline always return null

MuziburRahman, 2020-06-26:
Is this method yet to be implemented?

brettfo, 2020-06-26:
Calculating a proper bounding box for a spline is difficult and to avoid creating a box that's too big I'm not accounting for the curve's control points (see [here](https://github.com/ixmilia/dxf/blob/41b1e6f0454140d49d95e292b0e4f5b35d3c7cea/src/IxMilia.Dxf/Entities/DxfSpline.cs#L53-L58)).

I don't want to force the control points into the calculation in this library, but if you needed something more workable you could implement it on your end like so:

``` csharp
public static DxfBoundingBox? GetBoundingBoxForEntity(DxfEntity entity)
{
    switch (entity)
    {
        case DxfSpline spline:
            // custom implementation for splines; result will be overly large, but guaranteed to contain everything
            return DxfBoundingBox.FromPoints(spline.FitPoints.Concat(spline.ControlPoints.Select(c => c.Point)));
        default:
            // otherwise fall back to default behavior
            return entity.GetBoundingBox();
    }
}
```

nzain, 2020-06-26:
A spline cannot go beyond its control points, thus the bounding box of the control points definitely covers the spline. However, @brettfo is right: it _might_ be too large. How bad that is depends on your requirements and on the type of splines. Splines with many smooth control points may be very well bboxed that way. Few control points with strong directional changes, however, will produce an overly large bbox.

For Bézier curves and B-splines the following article is... great/lengthy/heavy to read. It talks about bounding boxes as well: https://pomax.github.io/bezierinfo/#boundingbox
Essentially, to calculate a precise bbox you have to find the roots, which may not be too complicated for quadratic and cubic curves. For higher degrees this is a _royal pain in the backside_ (his words) and most probably you give up or approximate. The degree in `DxfSpline` can be anything, I guess. I don't see a way for a generic correct implementation.

sampletext32, 2020-09-21:
So, I shouldn't rely on built-in ```GetBoundingBox``` for dxf files with splines?

brettfo, 2020-09-22:
> So, I shouldn't rely on built-in GetBoundingBox for dxf files with splines?

Correct. Since splines can be arbitrarily complicated, it's currently beyond what I have time (and let's be honest, the mathematical ability) to support.  If, however, I received a PR adding spline bounds that works in the general sense and I was convinced it was correct, (e.g., with adequate unit tests and a simplified explanation that I could grasp) I would gladly merge it in.

closed 2020-09-22 by brettfo

nzain, 2020-09-22:
Calculating the bbox around the control points will often provide a
_useful_ approximation. Plus: it's better than nothing 😉

2020, 19:58:

sampletext32, 2020-10-10:
@brettfo, actually there is a straight bruteforce-like way to determine spline's bounding box - follow its definition. If you separate spline in some-kind reasonable amount of points - you can simply build a bounding box from them. I have a working solution, but it's very memory dependant. Take a look [here is the link](https://gist.github.com/sampletext32/722175ce5a933a70ea547ad3196aff52) 
float t is normalized position along spline [from 0 to 1]

nzain, 2020-10-10:
The problem here is "reasonable amount of points": This depends on the
shape *and* length *and* the desired precision. Would you aim for a
millimeter precision on a 20km spline? If you take 100 samples of the
spline, you'll end up with a sample every 200m. If your spline is a
straight line, two points are sufficient. My point is: your solution is
another approximation, that might be ok for some users, but not necessarily
sufficient for all users.