ixmilia/dxf #136: DxfHatch Example

QbsCure, 2020-07-21:
Hello,
I can not understand how to create the Hatch.

For example, I have a triangle with three points through which I want to start Hatch.

DxfPoint pt1 = new DxfPoint (0, 0, 0)
DxfPoint pt2 = new DxfPoint (1, 1, 1)
DxfPoint pt3 = new DxfPoint (2, 2, 2)

How to set its boundaries and is it possible to set them for a 3D plane?

                 DxfLine ln1 = new DxfLine (pt1, pt2);
                 DxfLine ln2 = new DxfLine (pt2, pt3);
                 DxfLine ln3 = new DxfLine (pt3, pt1);

                DxfHatch hatch = new DxfHatch ();

brettfo, 2020-07-23:
`HATCH` objects are really complicated and this library can only do bare-bones experimental support for them.  That said, a hatch from a set of points could be constructed something like this:

``` csharp
var boundaryPath = new DxfHatch.NonPolylineBoundaryPath(DxfHatch.BoundaryPathType.Polyline);
boundaryPath.Edges.Add(new DxfHatch.LineBoundaryPathEdge() { StartPoint = pt1, EndPoint = pt2 });
boundaryPath.Edges.Add(new DxfHatch.LineBoundaryPathEdge() { StartPoint = pt2, EndPoint = pt3 });
boundaryPath.Edges.Add(new DxfHatch.LineBoundaryPathEdge() { StartPoint = pt3, EndPoint = pt1 });

var hatch = new DxfHatch();
hatch.PatternName = "SOLID";
hatch.BoundaryPaths.Add(boundaryPath);
```

As for a 3D plane, I have no idea if AutoCAD supports different z-values for a boundary; I've never tried it.  You can, however, have a hatch on something other than z=0 by setting the `Z` value of `pt1`, etc. to `0.0` and set `hatch.Elevation` to the desired z-value.

closed 2020-07-23 by brettfo

nzain, 2020-07-23:
Often 3D functionality is created by a flat 2D drawing and a normal vector
- confusing, yes. However, this worked for me with circles and polylines
and I have no idea about hatch.