ixmilia/dxf #138: XData hidden on some entities

zeehans, 2020-08-18:
I noticed that XData is hidden/ private on some entities such as DxfLine. I am not too familiar with the intricacies of the Dxf format, but don't think there is anything in the specs that would suggest that they ought to be hidden for some. So I am just wondering why they decision was made to make them private? Is it possible to make them public via the EntitiesSpec.xml?

zeehans, 2020-08-18:
Apologies - just found the HasXData Attribute. I suppose that would be the right one to set?

brettfo, 2020-08-18:
The spec doesn't mention XData in the  ["Common Group Codes for Entities"](http://help.autodesk.com/view/OARX/2020/ENU/?guid=GUID-3610039E-27D1-4E23-B6D3-7E60B22BB5BD) section and XData isn't listed for [`LINE`](http://help.autodesk.com/view/OARX/2020/ENU/?guid=GUID-FCEF5726-53AE-4C43-B4EA-C84EB8686A66), either, but it _is_ listed for [`MTEXT`](http://help.autodesk.com/view/OARX/2020/ENU/?guid=GUID-5E5DB93B-F8D3-4433-ADF7-E92E250D2BAB) (and I've enabled it for `MTEXT`) which makes me think that XData is only valid on the entities that explicitly allow it.

Is there specific XData that you expect to use on `DxfLine`, etc.?  Or do you have a source that states that XData should be available (or even better, a DXF file produced by AutoCAD)?  I honestly have no idea and just assumed that if the spec didn't mention it then it wasn't allowed, but I could very well be wrong.

zeehans, 2020-08-18:
I have quite a few DXF files but do not know if they were produced by AutoCAD. I can only say that we used to use a commercial DXF/CAD library (and are evaluating IxMilia right now as you guys are supporting a lot of formats) and that one does extract the XData for DxfLine, DxfCircle and DxfArc entities. There is another DxfReader on github (netDxf) that extracts it also. I also do not know if it's correct - I just assumed that any entity can have XData (which from my perspective would make sense). We use it to store some custom information on DxfLines that is specific to our use case. I can provide a file or two if you are interested. 

So unfortunately I do not have a source either, but I think it is no problem to make this opt-in via the HasXData="true" attribute in the specs file. I quite like that approach via the generator by the way! Nicely done!

brettfo, 2020-08-19:
The fact that other tools expose XData is good enough for me.  I've created a new branch `xdata` that exposes it on all entities and objects.  If you get a chance, can you give it a try?  If it works for you I'll merge it back to `main`.

zeehans, 2020-08-19:
I just ran a quick test on two dxf files and it looks good to me.

closed 2020-08-19 by zeehans

brettfo, 2020-08-19:
Glad to hear it works!  I've merged the changes back to `main`.