ixmilia/dxf #139: DxfCircle and DxfArc Center vector in local coordinates

zeehans, 2020-08-24:
Apparently this is a peculiarity of Dxf, but for circles and arcs, the position vector (e.g. center) is given in local coordinates whereas for e.g. ellipses it is in world coordinates. I transform the coordinates on the caller side, but am wondering if that shouldn't really be done during the reading process to have all entities in the same coordinate system.

brettfo, 2020-08-26:
I had a similar philosophical battle regarding `ARC` and `ELLIPSE` entities; [arcs](http://help.autodesk.com/view/OARX/2020/ENU/?guid=GUID-0B14D8F1-0EBA-44BF-9108-57D8CE614BC8) specify their start and end angles in degrees while [ellipses](http://help.autodesk.com/view/OARX/2020/ENU/?guid=GUID-107CB04F-AD4D-4D2F-8EC9-AC90888063AB) use radians and unless you know that, it's very odd dealing with finding the start and end points of the arcs.  On that front I decided to avoid, wherever possible, any "magic" when reading/writing entities to match as closely as possible the file on disk.

Using that argument I'd like to leave the `CIRCLE` (and its subclass `ARC`) center points as they are.  I'm not opposed, however, to having a method on `DxfViewport` that can return a modified entity, possibly similar to this:

``` c#
public class DxfViewport
{
    public DxfEntity TransformEntity(DxfEntity entity)
    {
        switch (entity)
        {
            case DxfCircle circle:
                return ModifyCenterPoint(circle);
            default:
                // other entities are unchanged
                return entity;
        }
    }

    private DxfCircle ModifyCenterPoint(DxfCircle circle)
    {
        // TODO
    }
}
```

What are your thoughts?

nzain, 2020-08-26:
I agree with @brettfo - keep original values (DXF is a mess, but adding magic will confuse those that expect the messy values). It's terrible, but you better know the DXF trickery because other apps will open your ixmilia-generated files sooner or later.

Maybe the new method name could be more concise - what _transform_ is it? From local to viewport / world coordinates?

zeehans, 2020-08-27:
I can only refer to other Dxf libraries again that I have used (netDxf and a commercial one) and both transform said entities after they have been read and before they are made available to the caller. netDxf deals with it similarly to what you both are suggesting.

brettfo, 2020-08-27:
I'm still leaning towards not doing any magic when reading or writing the file.

However, doing a bit more reading in the spec around [Object Coordinate Systems](http://help.autodesk.com/view/OARX/2020/ENU/?guid=GUID-D99F1509-E4E4-47A3-8691-92EA07DC88F5), I think it may be interesting to add a method to the `DxfEntity` base class: `public virtual DxfTransform GetTransform() => null;` that will be overridden for entities like `DxfCircle`: `public override DxfTransform GetTransform() => /* something meaningful */;`, where `DxfTransform` is a new class with a method `public DxfPoint Transform(DxfPoint point)` and a similar override for `DxfVector`.

What are your thoughts on this approach?  The OCS page in the spec mentions the [Arbitrary Axis Algorithm](http://help.autodesk.com/view/OARX/2020/ENU/?guid=GUID-E19E5B42-0CC7-4EBA-B29F-5E1D595149EE) which I've already implemented [here](https://github.com/ixmilia/dxf/blob/3113690c0a2d506928936dfbcd5db0f1971e8e92/src/IxMilia.Dxf/DxfVector.cs#L140-L153), so we can draw upon that.

zeehans, 2020-08-28:
Yes I implemented that algorithm on the caller side also as I wasn't aware that you had already done it. I am also OK with your suggested approach.

closed 2020-08-28 by zeehans