ixmilia/dxf #140: XData as collection (rather than single object)

zeehans, 2020-08-27:
If I understand the spec correctly (and I assume so as other libraries do that also), should XData be a collection or dictionary. For each entity it should be possible to have multiple <AppId, Extended data> tuples/pairs. The relevant section in the [spec ](http://help.autodesk.com/view/OARX/2020/ENU/?guid=GUID-A2A628B0-3699-4740-A215-C560E7242F63)reads:

[...]Extended data is grouped by registered application name. Each registered application group begins with a 1001 group code, with the application name as the string value. Registered application names correspond to APPID symbol table entries.[...]

brettfo, 2020-08-27:
I think you're right.  After re-reading the spec my understanding is that a single XData item is a `1001/<name>` code pair and either a single value (e.g., `1040/1.0`) or a list (e.g., `1002/{`, ... `1002/}`) and that any entity can have multiple XData items so they should be gathered into a `IDictionary<string, XDataItem>`.

With that my understanding is that the following code pairs...

```
1001
group_name_1
1040
1.0
1001
group_name_2
1002
{
1010
11.0
1020
22.0
1030
33.0
1002
}
```

...should be represented by this structure (using JSON because it's convenient for visualization):

``` json
{
  "group_name_1": [
    [1040, 1.0]
  ],
  "group_name_2": [
    [1010, 11.0],
    [1020, 22.0],
    [1030, 33.0]
  ]
}
```

Is that how you read it to be?  And yes, I would be interested in seeing a file that has multiple XData items on an entity just so I can verify it.  You can either paste the relevant code pairs in this issue, or if you'd like to keep it private you can email it directly to me at brett.forsgren &#64; outlook.com and I'll delete it as soon as I've confirmed the structure.

zeehans, 2020-08-28:
Yes that's how I understand it as well. Here a sample file: [xdatadictionary.dxf.txt](https://github.com/ixmilia/dxf/files/5140075/xdatadictionary.dxf.txt)

brettfo, 2020-08-29:
I've added a branch `xdata` with some initial work on this, can you give it a shot and see if you can access the XData you need?

It's not quite ready to merge to `main` yet, but I want to make sure I'm going down the right path.

zeehans, 2020-09-02:
Apologies for the absence, but am currently on holiday. I will check it out as soon as I have some time.

zeehans, 2020-09-02:
Just checked it out and can access the XData I need. I think that's how it is supposed to work. Thanks for fixing that so quickly.

brettfo, 2020-09-03:
Glad it works.  I've merged those changes back to `main` via commits b875f236baa06a612fed7e09c7070a975c0daef8 and 75d6db8a88b819bb49f21aea7d1a736f718cf96f.

closed 2020-09-03 by brettfo