ixmilia/dxf #142: Question: DxfColor ByBlock, where to find the block color?

Imagine-Programming, 2020-10-06:
Hello,

I'm trying to understand ixmilia/dxf in conjunction with the DXF standard, but I'm stuck on one thing; ByBlock colors. If I would like to get the applied DxfColor instance for any entity, how would I do so for `IsByBlock == true`?

Is there a way of accurately getting the color that is applied to an entity, regardless of it being ByLayer or ByBlock?

Thank you.

brettfo, 2020-10-07:
### Short answer
It's really complicated and you'll have to track layer and block colors while you're iterating through the file.  Example at bottom.

### Long answer
The DXF spec is really loose in what it technically allows, but AutoCAD itself is really strict in what it requires.  My middle-ground approach was to try to accept any DXF, even if it's something that AutoCAD would crash on.  The downside is that the burden of tracking a lot of stuff falls on you.  As a simple example, entities have a `string` property that states what layer they belong on.  Consider the following `LINE` on layer `Electrical` and a color value of `BYLAYER` (256):

```
  0
LINE
  8
Electrical
 62
256
...(other stuff about the line)
```

This seems easy enough, now let's find the layer named "Electrical" and get its color:

``` csharp
var layerColor = dxfFile.Layers.Single(l => l.Name == "Electrical").Color;
```

That line may throw an exception because the DXF file isn't required to have a layer by that name.

### Example
If you'd like to "properly" track block and layer colors, you'd have to do something like this:

``` csharp
foreach (var layer in dxfFile.Layers)
{
    // note that this will never bring in entities that don't match an existing layer, like "Electrical" above; what to do?
    foreach (var entity in dxfFile.Entities.Where(e => e.Layer == layer.Name))
    {
        var entityColor = entity.Color;
        if (entityColor.IsByLayer)
        {
            entityColor = layer.Color;
        }
        if (entityColor.IsByBlock)
        {
            // we're not in a block here so BYBLOCK doesn't make sense; what to do?
        }

        if (entity is DxfInsert insert)
        {
            // this is a block being inserted
            var block = dxfFile.Blocks.SingleOrDefault(b => b.Name == insert.Name);
            if (block != null)
            {
                foreach (var blockEntity in block.Entities)
                {
                    var blockEntityColor = blockEntity.Color;
                    if (blockEntityColor.IsByLayer)
                    {
                        blockEntityColor = layer.Color;
                    }
                    if (blockEntityColor.IsByBlock)
                    {
                        blockEntityColor = insert.Color;
                    }

                    // do something with `blockEntityColor`
                }
            }
            else
            {
                // the expected block definition wasn't found; what to do?
            }
        }
        else
        {
            // another entity type (e.g., CIRCLE or LINE), do something with `entityColor`
        }
    }
}
```

Even with all that said, color in a DXF file isn't an actual color, it's a color index number, and you can redefine that to be whatever you like; there's no guarantee that what you see as blue will show up on my machine as blue. **But**, it's not all bad in that regard; the AutoCAD color palette is pretty well known so there's a `.ToRGB()` method on `DxfColor` that'll get you an `int` from the default color palette.

Imagine-Programming, 2020-10-07:
Thank you for the clear explanation and example. This information is quite useful to me, as I don't really need to follow the "expected colors". In my implementation, I provide a method of replacing color indices in DXF files for certain entities with another index, regardless of what they might look like. Thank you.

closed 2020-10-07 by Imagine-Programming