ixmilia/dxf #143: Exporting Points

Happe79, 2020-10-19:
Hello,

I have a little bit problems to export Points into a DXF-File. I used that code:

   private void WritePoint(DxfFile dxfFile, string sLayer)
        {
            MyPoint p = new MyPoint(10.0,10.0);
            
            DxfModelPoint entityObj = new DxfModelPoint(new DxfPoint(p.x, p.y, 0.0));           
            entityObj.Layer = sLayer;
            entityObj.Color = DxfColor.ByBlock;
            entityObj.Thickness = 10;
      
            dxfFile.Entities.Add(entityObj);
        }

In the exported file the point seemes to be there but the viewer does not show it. Any ideas?

THX in advance.

Happe
[inV10.txt](https://github.com/ixmilia/dxf/files/5401850/inV10.txt)

brettfo, 2020-10-19:
The `DxfModelPoint` appears to be written to the file correctly.  What viewer are you using?

Happe79, 2020-10-20:
I am using E-Drawing from Solidworks. I guess it is more viewer problem. I was just wondering if there is a easy way to set a symbol for the points ($PDMODE).

Thank you for the support.

Happe

brettfo, 2020-10-20:
Thanks for the pointers.  I was able to get points to display by setting the point display mode to `2` and the size to `1.0` (or some other value depending on the scale of your drawing).

You can set those values with the specific properties like this:

``` csharp
DxfFile file = new DxfFile();
// ...
file.Header.PointDisplayMode = 2;
file.Header.PointDisplaySize = 1.0;
```

or you can use the header variables like so:

``` csharp
DxfFile file = new DxfFile();
// ...
file.Header["$PDMODE"] = 2;
file.Header["$PDSIZE"] = 1.0;
```

I haven't experimented much more, but other values for `PointDisplayMode` should yield different styles.

Happe79, 2020-10-22:
Perfect. It is working :).

Thanks again.

closed 2020-10-22 by Happe79