ixmilia/dxf #148: StreamExtensions.GetString Improvement

sampletext32, 2020-11-20:
https://github.com/ixmilia/dxf/blob/a909a05d80fb335e7d08f85289cffe572516edbe/src/IxMilia.Dxf/Extensions/StreamExtensions.cs#L41-L50

I think this piece of code is valid only for ```Encoding.ASCII```
So I suggest a removal of this string construction. And a replace with ```Encoding.ASCII.GetString```

sampletext32, 2020-11-20:
And a more heave suggestion - just use ```StreamReader```

nzain, 2020-11-20:
This code gets executed only, if `encoding == null` as a fallback. Sounds ok to me? Reading strings from a generic `Stream` is cumbersome, indeed - but you don't want to construct a `StreamReader` to read a few bytes only. The `StreamReader` might consume more bytes than intended, bytes that you have to handle separately. As far as I can remember, the DXF specification is based around ASCII for 99% and you can't simply use any well known encoding (would be awesome, but DXF is totally weird in this aspect). I might confused it with PDF, though.

brettfo, 2020-11-20:
This byte-by-byte method is necessary and fails if Encoding.ASCII is used because it can't properly handle byte-order-marks which some DXF libraries mistakenly omit at the start of the file.  We need the raw string so we can special-case the BOM here: https://github.com/ixmilia/dxf/blob/a909a05d80fb335e7d08f85289cffe572516edbe/src/IxMilia.Dxf/DxfFile.cs#L224-L228

closed 2020-11-20 by brettfo