ixmilia/dxf #150: Exception when executing Loadfile call on some DXF files

uzername, 2020-11-23:
Hello. I am opening one dxf file and get such exception:

```
   at System.Linq.Enumerable.Last[TSource](IEnumerable`1 source)
   at IxMilia.Dxf.Entities.DxfHatch.NonPolylineBoundaryPath.TrySetPair(DxfCodePair pair) in D:\a\dxf\dxf\src\IxMilia.Dxf\Entities\DxfHatch.BoundaryPath.cs:line 115
   at IxMilia.Dxf.Entities.DxfHatch.TrySetPair(DxfCodePair pair) in D:\a\dxf\dxf\src\IxMilia.Dxf\Entities\DxfHatch.cs:line 27
   at IxMilia.Dxf.Entities.DxfEntity.PopulateFromBuffer(DxfCodePairBufferReader buffer) in D:\a\dxf\dxf\src\IxMilia.Dxf\Entities\DxfEntity.cs:line 309
   at IxMilia.Dxf.Entities.DxfEntity.FromBuffer(DxfCodePairBufferReader buffer) in D:\a\dxf\dxf\src\IxMilia.Dxf\Entities\Generated\DxfEntityGenerated.cs:line 499
   at IxMilia.Dxf.Blocks.DxfBlock.FromBuffer(DxfCodePairBufferReader buffer) in D:\a\dxf\dxf\src\IxMilia.Dxf\Blocks\DxfBlock.cs:line 213
   at IxMilia.Dxf.Sections.DxfBlocksSection.BlocksSectionFromBuffer(DxfCodePairBufferReader buffer) in D:\a\dxf\dxf\src\IxMilia.Dxf\Sections\DxfBlocksSection.cs:line 79
   at IxMilia.Dxf.Sections.DxfSection.FromBuffer(DxfCodePairBufferReader buffer, DxfFile parentFile, DxfAcadVersion version) in D:\a\dxf\dxf\src\IxMilia.Dxf\Sections\DxfSection.cs:line 60
   at IxMilia.Dxf.DxfFile.LoadFromReader(IDxfCodePairReader reader) in D:\a\dxf\dxf\src\IxMilia.Dxf\DxfFile.cs:line 274
   at IxMilia.Dxf.DxfFile.Load(Stream stream, Encoding defaultEncoding) in D:\a\dxf\dxf\src\IxMilia.Dxf\DxfFile.cs:line 207
   at IxMilia.Dxf.DxfFile.Load(Stream stream) in D:\a\dxf\dxf\src\IxMilia.Dxf\DxfFile.cs:line 194
   at DXFRendering.LOGICAL.DxfReadWrapper.processDxfFile(String in_obtainedFileName) in 
   DxfReadWrapper.cs:line 49
   at DXFRenderingBitmap.Form1.handleFile(String in_PathToDXF) in 
   Form1.cs:line 83
```

Message is : `Sequence contains no elements`

uzername, 2020-11-23:
File did not contain hatches

uzername, 2020-11-23:
Happened on several other files. This time file contained hatch, but inside block; even if that block was not INSERTed to *Model_space.

brettfo, 2020-11-23:
Can you attach a sample file so I can play around with it?  If you don't want the file to be public you can email it directly to me at `brett.forsgren` `@` `outlook.com` and I'll delete it as soon as I'm done.

uzername, 2021-01-16:
I think, it should be fixed in https://github.com/ixmilia/dxf/commit/602c6a7634713d4191795464a74dce452a6d7cac . I close issue now. Looking forward to check the solution with new version of Nuget package.

closed 2021-01-16 by uzername

uzername, 2021-01-25:
Solved in 0.8.0 Nuget package. works good.