ixmilia/dxf #151: Get Area

progLks, 2020-11-25:
Excuse me, I started using this package, and i'd like to know, can i get area of object or layer?

brettfo, 2020-11-25:
### Short Version

This library is only for reading and writing DXF files and exposing the information available.  Calculating the area of something falls into the domain of the CAD software you're using, independent of what file format you're using, whether it's DXF, DWG, IGES, etc.

### Long Version

It really depends on what you mean by "object or layer".  To get the area of something you first need an enclosed area.  The DXF spec really only has 4 enclosed entities which are exposed in this library as `DxfPolyline`, `DxfLwPolyline`, `DxfCircle`, and `DxfEllipse` (but only if `StartParameter`/`EndParameter` is 0/2π).  Finding the areas of circles and ellipses is simple, so I'll skip those so now I'll turn to `DxfPolyline`/`DxfLwPolyline`.  If the polyline is closed and all segments are straight lines (e.g., every vertex has a `Bulge` of 0.0), then it's pretty straight forward to find algorithms online to calculate the area inside a polygon.  It starts to get really messy, however, if some of those polyline segments are curves and the math involved there is beyond my ability and also beyond the scope of this library.

Taking the complexity up a notch, may CAD software suites allow you to define an area by clicking a single point and it'll then discover what lines and curves bound it.  From there it then does the same polygon area calculation discussed above.  That feature, though, is also beyond the scope of this library as it's not unique to the DXF file format.

Hope this helps.

progLks, 2020-11-26:
Thank you very much, I tried your solition and it helped me.

closed 2020-11-26 by progLks