ixmilia/dxf #152: DxfAppId with empty string as name

zeehans, 2020-12-14:
We have Dxf files from customers where the DxfLine entities have AppIds with empty strings as names. At instantiation of DxfAppId an error is thrown saying that the parameter value name must not be empty. I am not sure what the DXF spec says here but in practice this is often the case and a few of our customers' DXF files do not set the AppIds. As their CAD system allows them to save those files, I suppose that is permissible. Can we remove that restriction upstream?

brettfo, 2020-12-14:
Table items (e.g., AppIds, Layers, etc.) with empty names have been a source of bugs in the past (see issue #133), but I'd like to support consuming as many files as possible.  Can you get a file with an empty AppId name, as generated by your customers' CAD systems, to share with me?  If you don't want to share it publicly you can email me directly at `brett.forsgren` `@` `outlook.com` and I won't share the file with anyone else and I'll delete it as soon as I'm done with it.

zeehans, 2020-12-15:
@brettfo sure no problem. File attached.
[TEST3.dxf.txt](https://github.com/ixmilia/dxf/files/5694394/TEST3.dxf.txt)

brettfo, 2020-12-15:
The provided file doesn't have any AppId items present and I'm able to open it without issue using the latest code from `main`.  Do you have another file that fails to open because there's an AppId with a blank name?

zeehans, 2020-12-16:
Apologies - you are right. It wasnt the Dxf that triggered this but a setting in out application. However, I still think an empty AppId is valid. Some background: we manufacture metal bending machines and our customers' Dxfs are created from a variety of vendors. Each vendor uses their own definitions to identify certain lines that are relevant to create a 3D representation of the final product. 

These definitions are written into the lines' XDATA. The extended data's AppIds then describe eg the outer/ inner contours (ie holes or protrusions on/in the metal sheet) and bending zones (extended data itself describes things such as bending angles, radii and bend shortenings). With that information we can develop a 3D representation from the DXF 2D one. On top of the vendors' definitions we also support our customers' own ones, however the customer has to enter their AppId definitions into our application. 

That particular customer specified the AppId for bend lines to be empty, as they have not entered the information necessary to create the 3D representation at CAD design time. Our software now reads in the Dxf, does not find the information necessary and guides the customer through the process of defining contours and bend lines. When they choose a line in the Dxf to be a bend line, we create the extended data using the AppId the customer configured (whose Name property had been set to be empty). At that point the exception is thrown and we cannot instantiate the DxfAppId object. This also works with the old (commercial) Dxf library we used, so I think that should be OK.

brettfo, 2020-12-16:
If I'm understanding your scenario correctly, only AppId needs to support a blank name, not the other table items (e.g., LAYER, DIMSTYLE, etc.)?

If the above statement is correct, I'd still like to have a sample file with an unnamed AppId as generated by the other software so I can confirm I'm doing the correct thing and not working blind.

closed 2022-12-03 by brettfo