ixmilia/dxf #153: Special character like ö,ü, ß not working

Happe79, 2021-01-11:
Hello,

I have some problems to load a dxf file correctly which contain german "Umlaute" like "ö,ü,ß" . I tried different ways but I did not get it.

[image]

I tried different ways to load the file via 

DxfFile dxfFile;
using (FileStream fs = new FileStream(sFilename, FileMode.Open, FileAccess.Read, FileShare.ReadWrite))
{
// I did some preperations before here like converting to an UTF8 byte array, but no success
    dxfFile = DxfFile.Load(fs);
}

.
.
.
private void ReadText(DxfText text)
{
"text.Value" always has the "?" character for special characters, e.g. "Aufma?datum" instead of "Aufmaßdatum"
}

Thank you in advance.

I included an example file.

[umlaute.txt](https://github.com/ixmilia/dxf/files/5796792/umlaute.txt)

brettfo, 2021-01-11:
Thanks for finding and reporting this issue.  I explored a bit and I currently don't handle non-Unicode files (R2004 and previous) correctly.  I'll work on a fix for this shortly.

In the mean time you can use this workaround to properly use the 1252 code page, which _should_ be the default for most files:

``` csharp
using System.Text;
...
Encoding.RegisterProvider(CodePagesEncodingProvider.Instance); // this might only be needed on net4.x
var ansi1252 = Encoding.GetEncoding(1252);
var dxfFile = DxfFile.Load("path/to/file.dxf", ansi1252);
//                                             ^^^^^^^^ this forces the text to read correctly
```

brettfo, 2021-01-11:
I've fixed the issue in commit cd38b1220de8ac67b437dcfcc555fd46f15cde4b, so now you can simply read the file like you were doing in your original post.

closed 2021-01-11 by brettfo

Happe79, 2021-01-12:
Thank you very much for the fast response. 

I was able to make the changes inside my version which is from around one year ago.
I wanted to use the last version which I downloaded via Zip-File. But I am not able to compile this due to a lot of errors. E.g. DXFLine, DXFPolygone are not found. It is may just my fault. But I could not get it to work.

brettfo, 2021-01-12:
There are a lot of files that are generated before the build will succeed.  To get those to work (and to generate the missing `DxfLine`, etc.) you'll need to run `.\build-and-test.cmd` or `./build-and-test.sh`.  Once you've done that you can use `IxMilia.Dxf.sln` or `src\IxMilia.Dxf\IxMilia.Dxf.csproj` like normal.

Happe79, 2021-01-13:
Thank for that. My fault. :)