ixmilia/dxf #155: How to specify unit of an entity

MuziburRahman, 2021-01-31:
Let a circle of radius 4.0cm constructed with this code:
`new DxfCircle(new DxfPoint(0,0,0),  4.0)`

Now if I save this to a dxf file and then open it, it always show the radius 4.0 inch, not cm.
How do I specify in a `DxfFile` or in any other place what unit I intended to use?

MuziburRahman, 2021-01-31:
`dxf_file.Header.DefaultDrawingUnits = DxfUnits.Centimeters;` doesn't change the behavior

nzain, 2021-01-31:
I wrote a wiki entry about units. You need some ceremony and the right acad
version for units to work as intended.

MuziburRahman, 2021-01-31:
> I wrote a wiki entry about units. You need some ceremony and the right acad
> 
@nzain where can I find the wiki?

MuziburRahman, 2021-02-01:
nevermind, found it

MuziburRahman, 2021-02-01:
@nzain I used this code:
```
var dxf_file = new DxfFile();
dxf_file.Header.Version = DxfAcadVersion.R14; // I also tried with R13
dxf_file.Header.DefaultDrawingUnits = DxfUnits.Centimeters;
dxf_file.Entities.Add(new DxfCircle(new DxfPoint(0,0,0), 4.0));
dxf_file.Save(..);
```
Still when I open the dxf in CATIA or SolidWorks, it shows a circle with 4 inch radius, not 4cm radius.
Is the code right? Should I use different headers?

nzain, 2021-02-01:
You are right, my "write" sample was incomplete. I extended the code and verified that the drawing unit shows up in QCad. I'm not 100% sure about the dimensions, which are tricky to use anyways. The following works for me:

```cs
    DxfFile dxf = new DxfFile();

    dxf.Header.Version = DxfAcadVersion.R2013; // default version does not support units
    dxf.Header.DrawingUnits = DxfDrawingUnits.Metric;
    dxf.Header.DefaultDrawingUnits = DxfUnits.Centimeters;
    dxf.Header.UnitFormat = DxfUnitFormat.Decimal;
    dxf.Header.UnitPrecision = 3;
    dxf.Header.DimensionUnitFormat = DxfUnitFormat.Decimal;
    dxf.Header.DimensionUnitToleranceDecimalPlaces = 3;

    // define a circle with 4 cm radius
    dxf.Entities.Add(new DxfCircle(new DxfPoint(0,0,0), 4.0))
    dxf.Save(...);
```

Please don't ask me, which header property is required to make it work. Furthermore, make sure that all your user's CAD applications accept this format. Especially AutoCad is very picky and intolerant.

For the sake of completeness, the wiki entry is here: [https://github.com/ixmilia/dxf/wiki/Units](https://github.com/ixmilia/dxf/wiki/Units)

MuziburRahman, 2021-02-05:
doesn't work. same problem. I am using Catia V5 to check the dxf file.
does it work in your end? If yes, what software are you using to open the dxf file?

nzain, 2021-02-05:
I've used QCad this time. You can check/modify units under edit > drawing settings (out of my head). My code worked for me, maybe your app needs some different combo of properties?

The free ODA Drawings Explorer (open design alliance) does not display units. I don't recommend the free DWG TrueView (Autodesk) either.

jpsantos-mf, 2021-04-23:
I've found the issue/solution. 

According to [this post](https://knowledge.autodesk.com/support/autocad/learn-explore/caas/CloudHelp/cloudhelp/2016/ENU/AutoCAD-Core/files/GUID-4BB25870-8BFF-478D-8C04-88457311AAF2-htm.html) and [this very useful PDF](https://images.autodesk.com/adsk/files/autocad_2012_pdf_dxf-reference_enu.pdf) we need to change $DIMALTF depending if we are in IMPERIAL or METRIC systems.

Adding ```dxf.Header.AlternateDimensioningScaleFactor = 0.0394;``` to the [previous example](https://github.com/ixmilia/dxf/issues/155#issuecomment-770630296) worked for me.

closed 2022-12-03 by brettfo